velvet-os/imagebuilder #186: growing partition to full sd card size fails to boot

Simbaclaws, 2024-01-03:
Hi, 

I'm trying to boot the kukui ubuntu jammy image on a chromebook ideapad 3, and it works perfectly fine if I leave the sd card unaltered.
But that only gives me 3gb of space, while my sdcard has 512gb of space....

I tried using gparted and parted to change the partition size of the btrfs partition, then mounting the partition with:
```sudo mount /dev/mmcblk0p4 /mnt```

And then using:
```sudo btrfs filesystem resize max /mnt```

And this seems to resize the btrfs filesystem completely.

However, when I now put the sd card in my chromebook, it simply beeps once when I try to press alt + ctrrl + U. And does not want to boot at all anymore.

Are there any special steps that need to be taken in order to grow the partition size of the btrfs root partition?

I'm doing this on another pc with linux installed.

hexdump0815, 2024-01-03:
just run /scripts/extend-rootfs.sh as root user when booted from the sd card - that should extend the root fs to the full size of the sd card. good luck

Simbaclaws, 2024-01-03:
thank you I didn't know there was a script for that, that does it....

I have another question:

I tried installing onto the emmc, which worked, but then I wanted to expand the root partition with another disk using btrfs... Is this something that is doable somehow?

I have a sd card of 500gb that I would like to use to expand the root filesystem...

Currently I'm getting a:
end kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block

because I had used butterfs to expand the root filesystem partition with another disk /dev/sda

Any ideas on what I can do to expand my sd card storage onto the root partition?

Simbaclaws, 2024-01-03:
I'm trying to move my root partition to my sd card, but I'm now figuring out that something inside the boot partition probably references /dev/mmcblk0p4, do you perhaps know what I need to alter to change this to /dev/sda1?

I already tried altering the /etc/fstab.

Simbaclaws, 2024-01-03:
Currently I decided to just mount my /dev/sda1 sd card on my user's home directory with /etc/fstab and then use flatpaks to install apps in my home dir.

hexdump0815, 2024-01-14:
@Simbaclaws - sorry, i saw this github issue just now - not sure if this is possible with btrfs (might be) - the rootfs is hardcoded as the partition the kernel booted from+3 in the kernels on the image (https://github.com/hexdump0815/linux-mainline-mediatek-mt81xx-kernel/blob/master/misc.cbm/misc/cmdline) ... maybe give creating your own kernel kpart image, maybe even with initrd a try following the idea from https://github.com/hexdump0815/imagebuilder/blob/e20a5dfbfa7105932941ec26bf7ff0e1cfcd8174/doc/install-to-emmc-with-luks-full-disk-encryption.txt#L250-L285 ... in the same dir are also some docs about building and installing own kernels which might be interesting to read

Simbaclaws, 2024-01-15:
thank you for the clearvoyant response.

For now I'm happy with /dev/sda1 being mounted on my home dir.

it already gives a lot of space to work with.

If I find my root partition to be lacking space, I'll try to build my own kernel based on those instructions.

Thank you so much for taking the time to do all of this. It brought life to my cheap chromebook :)