velvet-os/imagebuilder #187: Working on Jacuzzi (HP Chromebook 11MK G9 EE) using Kukui images for Jammy

EnterTheVoid-x86, 2024-01-09:
I can confirm that jammy works on jacuzzi on both usb and internal storage with all drivers working, with the only issues being present with X11. Wayland works great and has no issues whatsoever.

hexdump0815, 2024-01-14:
thanks for the success report, i have added it to the readme: https://github.com/hexdump0815/imagebuilder/commit/b8ebe14763708416a97b20654cc8cb8cd5738024 - the high pitch noise is a known problem, see: https://github.com/hexdump0815/imagebuilder/issues/65 - suspend/resume might also help to get rid of it and i did not hear it again yet with a v6.6 kernel (see info in the github issue)

EnterTheVoid-x86, 2024-01-16:
> thanks for the success report, i have added it to the readme: [b8ebe14](https://github.com/hexdump0815/imagebuilder/commit/b8ebe14763708416a97b20654cc8cb8cd5738024) - the high pitch noise is a known problem, see: #65 - suspend/resume might also help to get rid of it and i did not hear it again yet with a v6.6 kernel (see info in the github issue)

I'd just like to quickly note that this Chromebook isn't burnet and is actually esche, and it's named HP Chromebook 11MK G9 EE, not x360 11MK G9 EE. Although maybe you could change it to say HP Chromebook x360 11MK G3 EE/11MK G9 EE (burnet/esche) because the hardware is pretty much the exact same.

hexdump0815, 2024-01-16:
@EnterTheVoid-x86 - which dtb is it using? the output of "cat /proc/device-tree/model" would be interesting ... so far there is no dts for esche, so i guess that it really is burnet in the end then maybe? does it say esche at the dev mode bootscreen too?

hexdump0815, 2024-01-16:
thanks for checking and the update - i have fixed it in the readme now: https://github.com/hexdump0815/imagebuilder/commit/365159765669408c529a7396190157d21a77454c