velvet-os/imagebuilder #189: Is it possible to install Manjaro ARM using your kernel and modules?

EnterTheVoid-x86, 2024-01-21:
Title

hexdump0815, 2024-01-21:
a few more details would definitely be helpful - for arm chromebooks there is some info here: https://github.com/hexdump0815/imagebuilder/blob/main/doc/install-to-emmc-on-arm-chromebooks.md

EnterTheVoid-x86, 2024-01-29:
> a few more details would definitely be helpful - in principle that should be possible as a linux system is just some kernel and some filesystem, so taking the kernel plus modules and putting it into another filesystem would be possible - some more files (firmware, systemd scripts etc.) might be required as well for some systems

 what kinds of details? hw is the same as my previous issue, 11mk g9 ee (kukui, esche/jacuzzi). i mostly just don't know where to start BUT i think it could be possible using the kernel from your image and following guides from arch linux arm

aneeshlingala, 2024-02-10:
I tried archlinuxarm with the linux-chromebook kernel, and it worked great...
may upload working files this week. i'm not so sure that there is the same kernel in manjaro. but, I will run some experiments and get back to you.

aneeshlingala, 2024-02-10:
and also, I did test this on kukui, but on kappa
but it should work fine with esche as well...

linsyking, 2024-02-10:
> I tried archlinuxarm with the linux-chromebook kernel, and it worked great... may upload working files this week. i'm not so sure that there is the same kernel in manjaro. but, I will run some experiments and get back to you.

I tried to use the image "ARMv8 AArch64 Multi-platform" from https://archlinuxarm.org/about/downloads. It booted but the wifi doesn't work. It cannot find any wifi device.

EnterTheVoid-x86, 2024-02-21:
> and also, I did test this on kukui, but on kappa but it should work fine with esche as well...

The ALARM image does infact boot, but X11 and wayland do not work at all. I think it's missing graphics drivers

aneeshlingala, 2024-02-21:
I did get alarm working with wifi and Wayland, I just need to package it.

PD-Kerman, 2024-06-11:
> I tried archlinuxarm with the linux-chromebook kernel, and it worked great...
> 
> may upload working files this week. i'm not so sure that there is the same kernel in manjaro. but, I will run some experiments and get back to you.

Cool,wondering if I can have a try~ btw is there any guide on how to build such image with all kinds of different distributions? Thanks in advanced

linsyking, 2024-07-07:
I build a simple script to install archlinux on trogdor. Other devices should be similar.

https://github.com/linsyking/imagebuilder

aneeshlingala, 2024-07-07:
A few months ago I tried making a script like this for kukui.

PD-Kerman, 2024-07-07:
> I build a simple script to install archlinux on trogdor. Other devices should be similar.
> 
> https://github.com/linsyking/imagebuilder

wow cool, so that means for other devices we could just replace the

PD-Kerman, 2024-07-07:
> I build a simple script to install archlinux on trogdor. Other devices should be similar.
> 
> https://github.com/linsyking/imagebuilder

wow thats so cool, ive looked around the script a little bit, so that means we could just change the IMAGE_SRC var
(*seems like the rest of the cofde should be general/universal, and i dont see it used the file in the extra-file for 7180/7c next to the sxeipt in your repo~ i hope im guessibg right)
 in the run.sh script and replace it with our own kernel release for example like kukui and it will be all good to build image for an other arm chromebook right~? :D thats so cool

aneeshlingala, 2024-07-07:
Did you bypass the rootfs issue? Would be cool if you did!

linsyking, 2024-07-07:
> wow thats so cool, ive looked around the script a little bit, so that means we could just change the IMAGE_SRC var (*seems like the rest of the cofde should be general/universal, and i dont see it [...]

Previous version may not work, and I was working on a Readme now. You probably also need to move some device-specific firmware to `extra-files`.

panbed, 2024-07-14:
> I build a simple script to install archlinux on trogdor. Other devices should be similar.
> 
> https://github.com/linsyking/imagebuilder

Just used your script to install ALARM on my kukui/kappa, had make a few changes specific for kukui but everything seems to work pretty well so far! Thanks for making the script!

PD-Kerman, 2024-07-19:
> Just used your script to install ALARM on my kukui/kappa, had make a few changes specific for kukui but everything seems to work pretty well so far! Thanks for making the script!

:D that's so coool, I've got a kukui as well and I've been hoping for a archlinux image for kukui~ 
Would u care to share the image please~?

panbed, 2024-07-19:
> :D that's so coool, I've got a kukui as well and I've been hoping for a archlinux image for kukui~ Would u care to share the image please~?

Sure, give me a few.

PD-Kerman, 2024-07-19:
> > :D that's so coool, I've got a kukui as well and I've been hoping for a archlinux image for kukui~ Would u care to share the image please~?
> 
> 
> 
> Sure, give me a few.

Okey, thanks so much
And it should be better to maintain if we could have the modified script maybe uploaded as a repo as well~ :D

aneeshlingala, 2024-07-19:
> Take your time~ :)

Me too!

panbed, 2024-07-19:
> Okey, thanks so much Take your time~ :) And it should be better to maintain if we could have the modified script maybe uploaded as a repo as well~ Or simply sharing the script would be good as well :D Thanks again~

Here's the image:
https://github.com/panbed/imagebuilder/releases/download/kukui-kappa-image-19july2024/arch.img.xz

Ran into a bit of a snag while testing the image. Tried flashing it to a flash drive, it will boot like half way before erroring out on some BTRFS error and then it sets the filesystem as read-only. No idea what it is, I ran into the same issue when I was modifying the original script to run on my kappa on a flash drive but I got it working by just flashing directly to the eMMC as I didn't care for Chrome OS. Maybe my flash drive is dying or something, lol.

You can try building your own image by cloning my fork of linsyking's repo: https://github.com/panbed/imagebuilder. It's not much different, it just uses a different kernel and I think I changed the stuff in `extra-stuff` and maybe some other changes. Just follow the directions and it should work (given you have all the packages installed as well). 

Anyways, it's late and I need to go to work in a few hours. I might look into it again but no promises, lol.

PD-Kerman, 2024-07-19:
> > Okey, thanks so much Take your time~ :) And it should be better to maintain if we could have the modified script maybe uploaded as a repo as well~ Or simply sharing the script would be good as well :D Thanks again~
> 
> 
> 
> Here's the image:
> 
> https://github.com/panbed/imagebuilder/releases/download/kukui-kappa-image-19july2024/arch.img.xz
> 
> 
> 
> Ran into a bit of a snag while testing the image. Tried flashing it to a flash drive, it will boot like half way before erroring out on some BTRFS error and then it sets the filesystem as read-only. No idea what it is, I ran into the same issue when I was modifying the original script to run on my kappa on a flash drive but I got it working by just flashing directly to the eMMC as I didn't care for Chrome OS. Maybe my flash drive is dying or something, lol.
> 
> 
> 
> You can try building your own image by cloning my fork of linsyking's repo: https://github.com/panbed/imagebuilder. It's not much different, it just uses a different kernel and I think I changed the stuff in `extra-stuff` and maybe some other changes. Just follow the directions and it should work (given you have all the packages installed as well). 
> 
> 
> 
> Anyways, it's late and I need to go to work in a few hours. I might look into it again but no promises, lol.
> 
> 
> 
> (Also, obviously be careful when flashing stuff - I'm not responsible for borked systems lol)

Great! Thats so cool, btw the prebuilt image boots just fine~ but seems like **there's r username and password already set so I can't get into the system yet**, but it does boot fine and looks great.
And for some reason the script kept trying to connect some server in Florida, made it extremely slow to build. I'll see if I can change it later, I checked `run.sh` and there doesn't seem to be much code related
And I'm planning to maybe build the kernel for kukui with envkernel or manually from postmarketOS and install it on this system or maybe I'll see what I can do to make the script to do so.
Thanks for your work ! :D

panbed, 2024-07-19:
> > > Okey, thanks so much Take your time~ :) And it should be better to maintain if we could have the modified script maybe uploaded as a repo as well~ Or simply sharing the script would be good as well :D Thanks again~
> > 
> > 
> > Here's the image:
> > https://github.com/panbed/imagebuilder/releases/download/kukui-kappa-image-19july2024/arch.img.xz
> > Ran into a bit of a snag while testing the image. Tried flashing it to a flash drive, it will boot like half way before erroring out on some BTRFS error and then it sets the filesystem as read-only. No idea what it is, I ran into the same issue when I was modifying the original script to run on my kappa on a flash drive but I got it working by just flashing directly to the eMMC as I didn't care for Chrome OS. Maybe my flash drive is dying or something, lol.
> > You can try building your own image by cloning my fork of linsyking's repo: https://github.com/panbed/imagebuilder. It's not much different, it just uses a different kernel and I think I changed the stuff in `extra-stuff` and maybe some other changes. Just follow the directions and it should work (given you have all the packages installed as well).
> > Anyways, it's late and I need to go to work in a few hours. I might look into it again but no promises, lol.
> > (Also, obviously be careful when flashing stuff - I'm not responsible for borked systems lol)
> 
> Great! Thats so cool, btw the prebuilt image boots just fine~ but seems like **there's r username and password already set so I can't get into the system yet**, but it does boot fine and looks great. And for some reason the script kept trying to connect some server in Florida, made it extremely slow to build. I'll see if I can change it later, I checked `run.sh` and there doesn't seem to be much code related And I'm planning to maybe build the kernel for kukui with envkernel or manually from postmarketOS and install it on this system or maybe I'll see what I can do to make the script to do so, considering their kernel might be better maintained and got better performance :) Thanks for your work ! :D

Oh! Lol, the username and password should be `root` and `root` I believe. It's just the standard ALARM image. And my bad for the Florida thing, that's probably because I live in Florida and it uses those mirrors, lol.

I was also thinking of building a kernel since it seems like the kernel hexdump0815 built is a little outdated but I don't have much free time right now, lol.

Great that it works for you though, I was going insane trying to figure out why it didn't work with my flash drive last night.

aneeshlingala, 2024-07-19:
I had a non writavle error even on emmc when I built my own image a few
months ago :(

aneeshlingala, 2024-07-19:
Looks like a kernel issue

PD-Kerman, 2024-07-19:
> I had a non writavle error even on emmc when I built my own image a few
> 
> months ago :(
> 
> 
> 
> 
> 
> 
> > Okey, thanks so much Take your time~ :) And it should be better to
> 
> > maintain if we could have the modified script maybe uploaded as a repo as
> 
> > well~ Or simply sharing the script would be good as well :D Thanks again~
> 
> >
> 
> > Here's the image:
> 
> >
> 
> > https://github.com/panbed/imagebuilder/releases/download/kukui-kappa-image-19july2024/arch.img.xz
> 
> > Ran into a bit of a snag while testing the image. Tried flashing it to a
> 
> > flash drive, it will boot like half way before erroring out on some BTRFS
> 
> > error and then it sets the filesystem as read-only. No idea what it is, I
> 
> > ran into the same issue when I was modifying the original script to run on
> 
> > my kappa on a flash drive but I got it working by just flashing directly to
> 
> > the eMMC as I didn't care for Chrome OS. Maybe my flash drive is dying or
> 
> > something, lol.
> 
> > You can try building your own image by cloning my fork of linsyking's
> 
> > repo: https://github.com/panbed/imagebuilder. It's not much different, it
> 
> > just uses a different kernel and I think I changed the stuff in
> 
> > extra-stuff and maybe some other changes. Just follow the directions and
> 
> > it should work (given you have all the packages installed as well).
> 
> > Anyways, it's late and I need to go to work in a few hours. I might look
> 
> > into it again but no promises, lol.
> 
> > (Also, obviously be careful when flashing stuff - I'm not responsible for
> 
> > borked systems lol)
> 
> >
> 
> > Great! Thats so cool, btw the prebuilt image boots just fine~ but seems
> 
> > like *there's r username and password already set so I can't get into the
> 
> > system yet*, but it does boot fine and looks great. And for some reason
> 
> > the script kept trying to connect some server in Florida, made it extremely
> 
> > slow to build. I'll see if I can change it later, I checked run.sh and
> 
> > there doesn't seem to be much code related And I'm planning to maybe build
> 
> > the kernel for kukui with envkernel or manually from postmarketOS and
> 
> > install it on this system or maybe I'll see what I can do to make the
> 
> > script to do so, considering their kernel might be better maintained and
> 
> > got better performance :) Thanks for your work ! :D
> 
> >
> 
> > Oh! Lol, the username and password should be root and root I believe.
> 
> > It's just the standard ALARM image. And my bad for the Florida thing,
> 
> > that's probably because I live in Florida and it uses those mirrors, lol.
> 
> >
> 
> > I was also thinking of building a kernel since it seems like the kernel
> 
> > hexdump0815 built is a little outdated but I don't have much free time
> 
> > right now, lol.
> 
> >
> 
> > Great that it works for you though, I was going insane trying to figure
> 
> > out why it didn't work with my flash drive last night.

Well, it works like a charm here so far :D

PD-Kerman, 2024-07-19:
> > > > Okey, thanks so much Take your time~ :) And it should be better to maintain if we could have the modified script maybe uploaded as a repo as well~ Or simply sharing the script would be good as well :D Thanks again~
> 
> > > 
> 
> > > 
> 
> > > Here's the image:
> 
> > > https://github.com/panbed/imagebuilder/releases/download/kukui-kappa-image-19july2024/arch.img.xz
> 
> > > Ran into a bit of a snag while testing the image. Tried flashing it to a flash drive, it will boot like half way before erroring out on some BTRFS error and then it sets the filesystem as read-only. No idea what it is, I ran into the same issue when I was modifying the original script to run on my kappa on a flash drive but I got it working by just flashing directly to the eMMC as I didn't care for Chrome OS. Maybe my flash drive is dying or something, lol.
> 
> > > You can try building your own image by cloning my fork of linsyking's repo: https://github.com/panbed/imagebuilder. It's not much different, it just uses a different kernel and I think I changed the stuff in `extra-stuff` and maybe some other changes. Just follow the directions and it should work (given you have all the packages installed as well).
> 
> > > Anyways, it's late and I need to go to work in a few hours. I might look into it again but no promises, lol.
> 
> > > (Also, obviously be careful when flashing stuff - I'm not responsible for borked systems lol)
> 
> > 
> 
> > Great! Thats so cool, btw the prebuilt image boots just fine~ but seems like **there's r username and password already set so I can't get into the system yet**, but it does boot fine and looks great. And for some reason the script kept trying to connect some server in Florida, made it extremely slow to build. I'll see if I can change it later, I checked `run.sh` and there doesn't seem to be much code related And I'm planning to maybe build the kernel for kukui with envkernel or manually from postmarketOS and install it on this system or maybe I'll see what I can do to make the script to do so, considering their kernel might be better maintained and got better performance :) Thanks for your work ! :D
> 
> 
> 
> Oh! Lol, the username and password should be `root` and `root` I believe. It's just the standard ALARM image. And (maybe?) my bad for the Florida thing, it might be because I live in Florida and it used those mirrors, lol.
> 
> 
> 
> I was also thinking of building a kernel since it seems like the kernel hexdump0815 built is a little outdated but I don't have much free time right now, lol.
> 
> 
> 
> Great that it works for you though, I was going insane trying to figure out why it didn't work with my flash drive last night.

Okey got it~
lol it shouldn't be a big deal tho~ maybe ill go check which code caused that :) but simply using the image should be fine

Oh~ great lol, well I really recommend the kernel from postmarketOS lol, they've got all kinds of patches here:
https://gitlab.com/postmarketOS/pmaports/-/tree/master/device/community/linux-postmarketos-mediatek-mt8183?ref_type=heads

And simply applying those patches to the kernel src and use the preconfigured config file should be good I guess, I guess there r some docs on the pmOS wiki in the envkernel section which might be referable :D 
I'm a little bit busy these days, school stuff, I'll see what I can do later lol :)

PD-Kerman, 2024-07-19:
> I had a non writavle error even on emmc when I built my own image a few
> 
> months ago :(
> 
> 
> 
> 
> 
> 
> > Okey, thanks so much Take your time~ :) And it should be better to
> 
> > maintain if we could have the modified script maybe uploaded as a repo as
> 
> > well~ Or simply sharing the script would be good as well :D Thanks again~
> 
> >
> 
> > Here's the image:
> 
> >
> 
> > https://github.com/panbed/imagebuilder/releases/download/kukui-kappa-image-19july2024/arch.img.xz
> 
> > Ran into a bit of a snag while testing the image. Tried flashing it to a
> 
> > flash drive, it will boot like half way before erroring out on some BTRFS
> 
> > error and then it sets the filesystem as read-only. No idea what it is, I
> 
> > ran into the same issue when I was modifying the original script to run on
> 
> > my kappa on a flash drive but I got it working by just flashing directly to
> 
> > the eMMC as I didn't care for Chrome OS. Maybe my flash drive is dying or
> 
> > something, lol.
> 
> > You can try building your own image by cloning my fork of linsyking's
> 
> > repo: https://github.com/panbed/imagebuilder. It's not much different, it
> 
> > just uses a different kernel and I think I changed the stuff in
> 
> > extra-stuff and maybe some other changes. Just follow the directions and
> 
> > it should work (given you have all the packages installed as well).
> 
> > Anyways, it's late and I need to go to work in a few hours. I might look
> 
> > into it again but no promises, lol.
> 
> > (Also, obviously be careful when flashing stuff - I'm not responsible for
> 
> > borked systems lol)
> 
> >
> 
> > Great! Thats so cool, btw the prebuilt image boots just fine~ but seems
> 
> > like *there's r username and password already set so I can't get into the
> 
> > system yet*, but it does boot fine and looks great. And for some reason
> 
> > the script kept trying to connect some server in Florida, made it extremely
> 
> > slow to build. I'll see if I can change it later, I checked run.sh and
> 
> > there doesn't seem to be much code related And I'm planning to maybe build
> 
> > the kernel for kukui with envkernel or manually from postmarketOS and
> 
> > install it on this system or maybe I'll see what I can do to make the
> 
> > script to do so, considering their kernel might be better maintained and
> 
> > got better performance :) Thanks for your work ! :D
> 
> >
> 
> > Oh! Lol, the username and password should be root and root I believe.
> 
> > It's just the standard ALARM image. And my bad for the Florida thing,
> 
> > that's probably because I live in Florida and it uses those mirrors, lol.
> 
> >
> 
> > I was also thinking of building a kernel since it seems like the kernel
> 
> > hexdump0815 built is a little outdated but I don't have much free time
> 
> > right now, lol.
> 
> >
> 
> > Great that it works for you though, I was going insane trying to figure
> 
> > out why it didn't work with my flash drive last night.

Well, flashing into eMMC works fine so far, but we've got some kind of BTRFS warnings when booting from usb flash drive now :/

aneeshlingala, 2024-07-30:
I updated eballetbo's chromebook script to support kappa. this means we should be having fedora support. you can also build debian, but this repo already does that. i still need to test, might take a while. if anybody can test it, please clone this repo: [https://github.com/aneeshlingala/chromebooks](https://github.com/aneeshlingala/chromebooks), cd into it, and run 

`sudo ./chromebook-setup.sh do_everything --architecture=arm64 --storage=TARGET_DEV_NODE`

> make sure to replace "TARGET_DEV_NODE" as your target drive!
> this script will erase your drive.

linsyking, 2025-02-02:
Updates: I've updated [my repo](https://github.com/linsyking/imagebuilder) to support fedora because archlinux arm does not work well.
There are kernel building script, image building scripts and a pre-built image.