velvet-os/imagebuilder #190: Missing indented if in create-image.fs for chromebook_kukui

baybled, 2024-01-27:
execution stops and refers to line 402

```
# this currently would only be used by snapdragon_xyz systems and does not seem to work yet with
# the native arm64 grub included in debian and ubuntu so it is made to fail currently by
# specifying the not existing /boot/selected.dtb
# TODO: check if this is still the case
if [ "${UEFI64ARM}" = "true" ]; then
  chroot ${MOUNT_POINT} apt-get -yq install grub2-common grub-efi-arm64 grub-efi-arm64-bin
  chroot ${MOUNT_POINT} grub-install --target=arm64-efi /dev/loop0p1 --efi-directory=/boot/efi/ --boot-directory=/boot/ --no-nvram --no-bootsector --dtb=/boot/selected.dtb --removable
  # debian needs some extra steps to enable fallback boot sometimes required to boot from external media
  [ "$3" = "bookworm" ]; then
# lets not do this yet as long as that special grub version is still used via
# the systems extra-files/boot for the aarch64 images - maybe its time to check
# if the regular debian/ubuntu grub is meanwhile working well enough on aarch64?
#    chroot ${MOUNT_POINT} mkdir -p /boot/efi/EFI/BOOT
#    #chroot ${MOUNT_POINT} cp /boot/efi/EFI/debian/grubaa64.efi /boot/efi/EFI/BOOT/BOOTAA64.EFI
#    chroot ${MOUNT_POINT} cp /boot/efi/EFI/debian/grubaa64.efi /boot/efi/EFI/BOOT/grubaa64.efi
#    chroot ${MOUNT_POINT} cp /boot/efi/EFI/debian/fbaa64.efi /boot/efi/EFI/BOOT/fbaa64.efi
    chroot ${MOUNT_POINT} cp /usr/share/images/desktop-base/desktop-grub.png /boot/grub
  fi
fi
```

hexdump0815, 2024-01-27:
@baybled - oh thanks a lot for reporting this - it looks that was a mistake i seem to have introduced during the bulleye and focal support removal in https://github.com/hexdump0815/imagebuilder/commit/e20a5dfbfa7105932941ec26bf7ff0e1cfcd8174 ... i'll fix it immediately

hexdump0815, 2024-01-27:
fixed with https://github.com/hexdump0815/imagebuilder/commit/673537b9682e3c43ada64d37cd68234b0c6a7269

closed 2024-01-27 by hexdump0815