velvet-os/imagebuilder #193: Veyron to emmc installation: partitioning problem on Medion S2015

techorix, 2024-02-07:
I followed the guide on how to install to emmc here  https://github.com/hexdump0815/imagebuilder/blob/main/doc/install-to-emmc-on-arm-chromebooks.md but ran into problems witth the partitioning. Several devices simply could not be created so I had to recover the original chrome OS.

Here is the partition table from `fdisk -l` that I got working at the moment:
```
Disk /dev/mmcblk0: 14.68 GiB, 15758000128 bytes, 30777344 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 29F80876-EA43-124F-9630-7C9013CB0602

Device            Start      End  Sectors  Size Type
/dev/mmcblk0p1  8704000 30777295 22073296 10.5G Linux filesystem
/dev/mmcblk0p2    20480    53247    32768   16M ChromeOS kernel
/dev/mmcblk0p3  4509696  8703999  4194304    2G ChromeOS root fs
/dev/mmcblk0p4    53248    86015    32768   16M ChromeOS kernel
/dev/mmcblk0p5   315392  4509695  4194304    2G ChromeOS root fs
/dev/mmcblk0p6    16448    16448        1  512B ChromeOS kernel
/dev/mmcblk0p7    16449    16449        1  512B ChromeOS root fs
/dev/mmcblk0p8    86016   118783    32768   16M Linux filesystem
/dev/mmcblk0p9    16450    16450        1  512B ChromeOS reserved
/dev/mmcblk0p10   16451    16451        1  512B ChromeOS reserved
/dev/mmcblk0p11      64    16447    16384    8M unknown
/dev/mmcblk0p12  249856   315391    65536   32M EFI System

Partition table entries are not in disk order.

Disk /dev/mmcblk0boot0: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes

Disk /dev/mmcblk0boot1: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
```

As I have no idea how I should modify the sgdisk or cgpt cIommands accordingly, I was hoping, maybe the information above can help you recommend something to me. Maybe I should only delete certain partitions and write the kernels to the original kernel partitions p2, p4, p6?

hexdump0815, 2024-02-07:
it might be related to the problems described in the below issues and links - maybe have a look at them and see if anything of that helps - in general it should work on the medion s2015 as i have one myself and linux is running fine on it on emmc.

- https://github.com/hexdump0815/imagebuilder/blob/main/doc/important-information.md via https://github.com/hexdump0815/imagebuilder/issues/144
- https://github.com/hexdump0815/imagebuilder/issues/130#issuecomment-1830454965 and https://github.com/hexdump0815/imagebuilder/issues/16#issuecomment-1836208058
- most probably not really related: https://github.com/hexdump0815/imagebuilder/issues/110

here is some background about why the partitioning is so complicated on veyron:
https://www.osfc.io/2023/talks/when-boot-security-goes-wrong-the-story-of-the-ignoreme-gpt/

techorix, 2024-02-08:
Downgrading cgpt did the trick. It still felt a bit flaky due to all these error messages and warnings but worked like a charm, after all. Am very happy now. Thanks :)

closed 2024-02-08 by techorix