velvet-os/imagebuilder #194: How to add kernel arguments

FrostNovaHD, 2024-02-09:
Am running the latest trogdor img and idk how to add `PSI=1` normally you do that in grub but idk how on the img can any one tell me

FrostNovaHD, 2024-02-09:
Its a lenovo duet 3 btw

FrostNovaHD, 2024-02-09:
> Am running the latest trogdor img and idk how to add `PSI=1` normally you do that in grub but idk how on the img can any one tell me

This a fix for waydroid

hexdump0815, 2024-02-09:
the kernel cmdline is kind of built-in into the special image format used for the bootable kernels on chromebooks - i hope this: https://github.com/hexdump0815/imagebuilder/blob/e20a5dfbfa7105932941ec26bf7ff0e1cfcd8174/doc/install-to-emmc-with-luks-full-disk-encryption.txt#L235-L285 gives you an idea of how to adjust it and reassemble your own kernel image (then even with initrd) - you might have to adjust it a bit for your case - good luck and please let me know if you get it working this way