velvet-os/imagebuilder #195: Only 412 bytes of Space?

tech-with-mo, 2024-02-09:
So I have used Balena Etcher on my Windows PC to make a bootable Image. Worked. Booted via USB by pressing CTRL+U on Dev Mode Screen. Worked. Now I used youre write-to-emmc guide to write to the emmc storage, which also worked. But now for some Reasone I only have 412 bytes of Storage which is nowhere enough to install anything. I know that I have to expand the emmc? But I don´t know how. Any help is appreciated. (Lenovo Duet Chromebook 1 - Kukui. Jammy Image)

hexdump0815, 2024-02-09:
no sure which way you did choose to install to emmc, but maybe running /scripts/extend-ruutfs.sh is what you are looking for?

tech-with-mo, 2024-02-09:
Yup, exactly did that, worked just fine. Thank you tho!

closed 2024-02-10 by tech-with-mo