velvet-os/imagebuilder #198: External Displays

tech-with-mo, 2024-02-11:
Will external Displays ever work on kukui jammy img? Or is there a fix and I just missed it?

hexdump0815, 2024-02-11:
as far as i know the perople doing the mediatek mainlining have that topic on their radar, it is just not in mainline yet - so it will still take a bit of time, but at some time it will hopefully work - maybe there is something towards the end of the year?

one thing is important to note regarding external displays on kukui: the hardware is limited and can drive only displays up to around hd (1366x720) at 60hz, so large hi-res display will never work.

archisman-panigrahi, 2026-05-16:
It seems that the external display does not work in either of oak, kukui or gru chromebooks, while graphics driver are available for the last two. Is there any updates regarding this? In case it is not a missing GPU driver problem but some other problem, can we make any progress with AI?

hexdump0815, 2026-05-17:
@archisman-panigrahi - i tested oak hdmi myself at some point, but the hardware does not support hdmi via usb-c only hdmi via hdmi connector directly ... kukui should work ( https://github.com/hexdump0815/linux-mainline-mediatek-mt81xx-kernel/commit/dadebf45f07554034e96ea134c9699653eb37ad0 ) - maybe the documentation has not been updated yet.

archisman-panigrahi, 2026-05-17:
Sorry, I take back my comment about gru. My `gru` `dumo` (nearly same as scarlet) device works with USB-C to HDMI (perhaps it did not work one year ago with an older image, and I had a wrong impression). Thanks for the quick clarification.

I will also test and report back what is happening with scarlet.

Perhaps the only remaining problem is oak, but I will test the latest image and let you know.