velvet-os/imagebuilder #200: Disable command line boot

gufuskn, 2024-02-20:
How do I disable command line boot?
I'm on the Lenovo IdeaPad Flex 3 mt8183 Chromebook

hexdump0815, 2024-02-20:
i do not really understand what you want - can you please try to explain it once more in other and maybe a few more words?

gufuskn, 2024-02-20:
When the Chromebook turns on, there are 8 penguins at the top and processes that say “OK” and “FAILED”

gufuskn, 2024-02-20:
[image]

unixabg, 2024-02-20:
@gufuskn - you might try adding the boot param quiet . And you could consider adding the package plymouth

hexdump0815, 2024-02-21:
have a look at https://github.com/hexdump0815/imagebuilder/issues/194 maybe

gufuskn, 2024-02-22:
> have a look at #194 maybe

Alright, thanks

closed 2024-02-22 by gufuskn