velvet-os/imagebuilder #202: chromebook_kukui: lenovo 10e small issues

LukIsHere, 2024-03-14:
in general most things work:
audio, display, 3d acceleration, usb, attachable keyboard...

(at least on wayland)
but i have encountered some problems with automatic screen rotation
it is 90 degrees off
[image]
it also default 's to ^ this exact position in gdm3 and gnome 45 instead of landscape one
i was trying to fix it by modifying /usr/lib/udev/hwdb.d/60-sensor.hwdb
https://unix.stackexchange.com/questions/361472/changing-the-way-automatic-screen-rotation-works-in-gnome-shell
yet modifications don't seam to change anything
it's not really something i can't live without but if anyone know how to fix it would really appreciate some help

(at least on wayland)
also there is external display issue
usb-c to hdmi adapter doesn't work
the display name is detected correctly
but there is no video output on the screen
[image]
[image]
i have no idea if it is fixable

(at least on wayland)
last issue attaching or detecting keyboard on certain versions on gnome shell (iirc 44) can make the device completely ignore all inputs (touch,keyboard ...) after it happens restart is required if u want to still use the device
not sure what it's cause by but on gnome shell 45 the issue seam to not be the case anymore

if anyone has any solutions for them i would be grateful
it there is a need to check/test anything with the device would be happy to help
other than this this issue everything works perfectly well
for everyone who made it possible thank u❤️

LukIsHere, 2024-03-16:
fixed rotation issue

writing to /usr/lib/udev/hwdb.d/60-sensor.hwdb :
`
sensor:modalias:*
  ACCEL_MOUNT_MATRIX=0, 1, 0; -1, 0, 0; 0, 0, 1
`
**and then**
running
`
sudo systemd-hwdb update
sudo udevadm trigger
sudo reboot
`
makes the rotation work as intended

hexdump0815, 2024-03-17:
the non working external display is a known problem - the developers working on the mediatek mainline kernel are aware of it and i think some solution for that is on its way, but it might take quite a while until it works properly and ends up in mainline, i guess maybe in the second half of this year sometime ...

LukIsHere, 2024-03-17:
will it be announced somewhere so i can get a notification?

hexdump0815, 2024-03-23:
i think there will be no formal announcement about that, but whenever it will be supported in the images, i'll most probably update the kukui readme in the repo

LukIsHere, 2024-03-25:
also i have noticed that the high pitch noise is emitted from the device after being put into suspense while playing music or any audio at all
also noticed that the noise stops after leaving gnome by logging out and going back to gdm3 (or lightdm probably too)
so i want to ask if it would be possible to set it up so when putting into suspense it pauses all the audio playing currently?
i could probably get rid of the issue when it comes to listening to music and turning off the screen

hexdump0815, 2024-03-27:
this is a known problem with kukui devices: https://github.com/hexdump0815/imagebuilder/issues/65 - it seems to be better with a v6.6 kernel (see: https://github.com/hexdump0815/imagebuilder/issues/65#issuecomment-1883698383 ) - you might try to install this kernel: https://github.com/hexdump0815/linux-mainline-mediatek-mt81xx-kernel/releases/tag/6.6.9-stb-mt8%2B following those instructions: https://github.com/hexdump0815/imagebuilder/blob/main/doc/installing-a-newer-kernel.md ... good luck

LukIsHere, 2024-03-29:
thank u
i have managed to update the kelner
and i have one more question [postmarket page on kukui mainline](https://wiki.postmarketos.org/wiki/Google_Kukui_Chromebook_(google-kukui)) says that [patches for the kelner to add CSI camera support](https://lwn.net/ml/linux-media/7f8cb8c7-cbf2-40c9-ac8f-c3c9b97919d7@molgen.mpg.de/) has been opened since 2019, so would it be possible to add them to the mt81XX custom kelner?
i have little to no knowledge regarding kelner modifications so idk if it's possible or worth doing
it's not something very important to me since i don't really take that much pictures but it would still be nice addition

hexdump0815, 2024-03-29:
i'm aware of this old camera work, but i did not have the time yet to look into it - in the end someone needs to have a closer look to find out if it works or not - for time reasons that will not be me for the near future :)

LukIsHere, 2024-03-30:
i believe i run into issue with the way mesa is setup in the image
```
libEGL warning: MESA-LOADER: failed to open panfrost: /usr/lib/aarch64-linux-gnu/dri/panfrost_dri.so: nie można otworzyć pliku obiektu dzielonego: Nie ma takiego pliku ani katalogu (search paths /opt/mesa/lib/aarch64-linux-gnu/dri:/usr/lib/aarch64-linux-gnu/dri, suffix _dri)

libEGL warning: MESA-LOADER: failed to open zink: /usr/lib/aarch64-linux-gnu/dri/zink_dri.so: nie można otworzyć pliku obiektu dzielonego: Nie ma takiego pliku ani katalogu (search paths /opt/mesa/lib/aarch64-linux-gnu/dri:/usr/lib/aarch64-linux-gnu/dri, suffix _dri)

libEGL warning: MESA-LOADER: failed to open swrast: /usr/lib/aarch64-linux-gnu/dri/swrast_dri.so: nie można otworzyć pliku obiektu dzielonego: Nie ma takiego pliku ani katalogu (search paths /opt/mesa/lib/aarch64-linux-gnu/dri:/usr/lib/aarch64-linux-gnu/dri, suffix _dri)

qt.qpa.wayland: Failed to initialize EGL display 3001
qt.qpa.qgnomeplatform: Could not find color scheme  ""
QApplication: invalid style override 'adwaita-dark' passed, ignoring it.
	Available styles: Windows, Fusion
Loading gamepad mappings: /app/share/mcpelauncher/gamecontrollerdb.txt
Loading gamepad mappings: /app/share/mcpelauncher/gamecontrollerdb.txt

```
nie można otworzyć pliku obiektu dzielonego: Nie ma takiego pliku ani katalogu
translates to
cannot open shared object file, there is no such file or directory

programs have troubles opening /usr/lib/aarch64-linux-gnu/dri/panfrost_dri.so
running them as root solve the problem but we obviously don't want that

what would be the best solution to solve it (changing the owner comes to mind but idk if that's the right idea)

LukIsHere, 2024-06-18:
usi stylus has up/down swapped with left/right
haven't found a solution yet

K-J-VV, 2025-02-07:
> fixed rotation issue
> 
> writing to /usr/lib/udev/hwdb.d/60-sensor.hwdb :
> 
> ```
> sensor:modalias:*
>   ACCEL_MOUNT_MATRIX=0, 1, 0; -1, 0, 0; 0, 0, 1
> ```
> 
> **and then** running `sudo systemd-hwdb update && sudo udevadm trigger && sudo reboot` makes the rotation work as intended

Just completed the install guide & found this issue describing the same rotation issue on my lenovo 10e, but I am getting the following error after running `sudo udevadm trigger`

`sbs-12-000b: Failed to write 'change' to '/sys/devices/platform/soc/11012000.spi/spi_master/spi2/spi2.0/11012000.spi:cros-ec@0:i2c-tunnel/i2c-12/12-000b/power_supply/sbs-12-000b/uevent': Input/output error`

thenameisluk, 2025-02-07:
@K-J-VV just rebooting the device should also work

K-J-VV, 2025-02-07:
> [@K-J-VV](https://github.com/K-J-VV) just rebooting the device should also work

It did! Was actually just about to respond with that, haha. Seems to be working as expected now         still unsure about the error output from that command, unsure if it is important or not

thenameisluk, 2025-02-07:
@K-J-VV 
nope
that error is not really important
don't worry