velvet-os/imagebuilder #205: Question about corrupted/damaged firmware and bricked devices.

Dae-Hang-Sook, 2024-03-27:
My fennel14 is completely bricked because of firmware being messed up/corrupted/changed/updated etc.  Would the image provided for it help unbrick and fix the device and if so how would I go about doing so?

tech-with-mo, 2024-03-29:
What do you mean by bricked? Do you see the ChromeOS is missing or damaged screen? If so, unbricking would be possible. An image would be appreciated.

Dae-Hang-Sook, 2024-03-29:
> What do you mean by bricked? Do you see the ChromeOS is missing or damaged screen? If so, unbricking would be possible. An image would be appreciated.

No, there is nothing. The only thing that seems functions is the battery. (Indicated by the light and zero response from anything else)

tech-with-mo, 2024-03-29:
> > What do you mean by bricked? Do you see the ChromeOS is missing or damaged screen? If so, unbricking would be possible. An image would be appreciated.
> 
> No, there is nothing. The only thing that functions is the battery. (Indicated by the light and zero response from anything else)
> 
> To be fair if that's all I had seen I'd have fixed this a long long time ago.

Hmm, thats weird because that rather sounds like a Hardware issue rather than you messing up something Software wise. I suppose you look in a few forums or perhaps ask the manufacturer for further assistance. But if you ask me try powering on without the battery. If thats without look try a second battery. That pretty often resolves the Problem. But if it's the IdeaPad 3 you have there than I can comfirm that I had the same issue and I tried everything to fix it, without luck. I just bought a Duet 5 instead.

Dae-Hang-Sook, 2024-03-29:
> Hmm, thats weird because that rather sounds like a Hardware issue rather than you messing up something Software wise. I suppose you look in a few forums or perhaps ask the manufacturer for further [...]

Lenovo wouldn't like to hear that I was given a custom firmware for a device named mt8183 pumpkin (THIS IS NOT THE SAME DEVICE AS WAS FOUND OUT), asked multiple times to be sure it was safe with zero response, got told it should be generally safe, and it completely "brick" the device as they told me right after flashing it via flashrom. I figured maybe the img. File here had an original firmware file so I could use a suzy-q cable to flash it.  They insist it's bricked, but I'm pretty sure it's been completely killed and there's no way to fix it.

jenneron, 2024-04-04:
You can restore firmware with suzyq cable (if it actually allows you to flash SPI), but you need a backup made before. The firmware image shipped with a device contains mac address, maybe serial number and who knows what else device-specific. If there is no backup this information can't be restored

Dae-Hang-Sook, 2024-04-04:
> You can restore firmware with suzyq cable (if it actually allows you to flash SPI), but you need a backup made before. The firmware image shipped with a device may contain a mac address, possibly serial number and who knows what else device-specific. If there is no backup this information can't be restored

Well that's wonderful. Thanks a lot Google :D

(The only possible way to obtain a backup now would be pulling it out of a ChromeOS recovery drive but that's not entirely possible so there goes my device :,). )

jenneron, 2024-04-04:
> Well that's wonderful. Thanks a lot Google :D

why is it Google's fault?

jenneron, 2024-04-04:
> The only possible way to obtain a backup now would be pulling it out of a ChromeOS recovery drive

it is not there

> that's not entirely possible

we have recovery images, it is just that firmware image is not there

Dae-Hang-Sook, 2024-04-04:
> > Well that's wonderful. Thanks a lot Google :D
> 
> why is it Google's fault?

Don't get me wrong, the firmware problem is mostly my fault. That's not what I was meaning.

I do not have the tools available to extract files from a ChromeOS recovery drive. It's nearly impossible as it is for most people anyway and Lenovo most definitely would not give me anything to help. So I'll just have to get a new device instead of repairing mine.

jenneron, 2024-04-04:
> So I'll just have to get a new device instead of repairing mine.

if you get another fennel14 you may be able to read firmware from it using flashrom and then flash it to the old device using suzyq cable, but it would make those device-specific parts of firmware the same on both devices

Dae-Hang-Sook, 2024-04-04:
> did you flash a specific partition of SPI or the whole SPI storage?

And I'm not sure of the specifics of that. All I can say is I was instructed to use flashrom on the internal programmer (which I believe was the entire SPI chip)

But it's odd that the battery still functions.

Dae-Hang-Sook, 2024-04-04:
> > So I'll just have to get a new device instead of repairing mine.
> 
> if you get another fennel14 you may be able to dump firmware from it using flashrom and then flash it to the old device using suzyq cable, but it would make those device-specific parts of firmware the same on both devices

I'm good on that to be quite honest. The amount of issues I've had with the one I own as it is, I don't want to deal with it again. I'll just buy a more supported device instead and try my best to make sure it's not a Chromebook.

jenneron, 2024-04-04:
> to dump firmware from it using flashrom

you can use `flashrom -p linux_mtd -r backup.bin` for this

> I'm good on that to be quite honest. The amount of issues I've had with the one I own as it is, I don't want to deal with it again. I'll just buy a more supported device instead and try my best to make sure it's not a Chromebook.

i probably can't help with anything here, but i'm curious what was the initial problem?

jenneron, 2024-04-04:
> a more supported device

maintaining these devices in postmarketOS I haven't noticed any major issues e.g. with kernel, but I don't really know about its state in other projects right now

Dae-Hang-Sook, 2024-04-04:
> you can use `flashrom -p linux_mtd -r backup.bin` for this
>
>
> i probably can't help with anything here, but i'm curious what was the initial problem?

So initially I was trying to make my own version of U-Boot so I could use Linux instead of ChromeOS (as ChromeOS doesn't fit my needs but it was a Christmas present from my grandmother and I just couldn't have said no to that.) So, I was informed that there's something similar already built (mt8183_pumpkin) and should relatively not affect me much at all. It's the fact that after I flashed it I never turned my device off but upon waking the next morning to check it, it had shut off but wasn't dead. That's when I found out it infact had to be modified before install which I wasn't informed of. When asking all I got was "there's an unbricking guide in the docs"  but that's absolutely useless in my case. And yes, PostMarketOS did have a project for it but I can no longer find it anywhere on the website. (I'm assuming that's because there's a dedicated team who is trying to full write out an OS that will completely rid of ChromeOS)

Dae-Hang-Sook, 2024-04-04:
> So initially I was trying to make my own version of U-Boot so I could use Linux instead of ChromeOS (as ChromeOS doesn't fit my needs but it was a Christmas present from my grandmother and I just [...]

Also, with all ChromeOS devices I've owned, linux_mtd is not a programmer option nor is it accessible. Unsure as to why, that's just how it is. It's normally flashrom -p internal ****.  And so on

jenneron, 2024-04-04:
> So initially I was trying to make my own version of U-Boot so I could use Linux instead of ChromeOS (

well, we don't really have to use custom firmware for Linux, but U-Boot would certainly make it more convenient, so I hope we will have it in the future

> And yes, postmarketOS did have a project for it but I can no longer find it anywhere on the website.

do you mean images for fennel14? you can find those at https://postmarketos.org/download/ under "Google Kukui Chromebooks". they work on fennel14 and don't require custom firmware

> who is trying to full write out an OS that will completely rid of ChromeOS

you can replace chrome os with postmarketOS by flashing it to eMMC

>  that's just how it is. It's normally flashrom -p internal ****. And so on

ah, probably right. but I use `flashrom` not from chrome os, but from linux (postmarketOS) which uses upstream flashrom, so I use `-p linux_mtd`

jenneron, 2024-04-04:
> It's a Generic image

it is a generic image for kukui devices, yes, but there is no problem here

> how am I supposed to boot it? 

All the instructions are at https://wiki.postmarketos.org/wiki/Category:ChromeOS (assuming you can debrick your device)

> I can't even access the screen to press Ctrl + U to do it

you would need to enable booting from external storage with crossystem (again, assuming you can debrick the device)

> It's also very very buggy 

we may be unaware of some issues with the port that you have encountered, so you're really welcomed to report bugs. check if it is not reported at https://gitlab.com/postmarketOS/pmaports/-/issues/?sort=created_date&state=opened&not%5Blabel_name%5D%5B%5D=chromeos&first_page_size=20, then open a new issue. 

> I don't think very many people are doing anything with it either. 

we have 2 maintainers of kukui devices, but everything depends on many factors

> I tried to write out U-Boot for it

It wouldn't help with pmOS problems though. it would allow to boot other distributions, but problems are likely to come from upstream and happen there as well

> I could even have installed windows on it 

U-Boot does provide EFI firmware with SMBIOS, but it doesn't provide ACPI tables for OS to use. We don't use ACPI on these devices on Linux

jenneron, 2024-04-04:
> it would be helpful if someone was willing to get the file for me so I can do the rest by myself

I have a fennel14, but I currently don't live at the apartment, thus I don't have access to it right now. But I can get my hands on it within a week or maybe a few weeks. Although, I would share this only in private messages and I would discourage you from publishing firmware image taken from my device

You can find my contacts at https://wiki.postmarketos.org/wiki/User:Jenneron, please pick any and contact me privately

jenneron, 2024-04-04:
> within a week or maybe a few weeks

if someone pops into here and shares their image before that - it would be nice as well

Dae-Hang-Sook, 2024-04-04:
> it is a generic image for kukui devices, yes, but there is no problem here
>
>
> All the instructions are at https://wiki.postmarketos.org/wiki/Category:ChromeOS (assuming you can debrick your device) [...]

That's weird then. Ubuntu uses ACPI for the devices you install it on. I'm assuming that's it it's available as a device I had installed it on kept giving and ACPI error and now refuses to boot anything other than windows. (It's an old device)

Dae-Hang-Sook, 2024-04-04:
> I have a fennel14, but I currently don't live at the apartment, thus I don't have access to it right now. But I can get my hands on it within a week or maybe a few weeks. Although, I would share [...]

Will do. Thanks so much for you assistance as the last several people I tried just kept saying "check the docs" like it's going to help at all. Bunch of hopeless people I say. They rely too much on already existing information and don't try doing things for themselves.

jenneron, 2024-04-04:
> That's weird then. Ubuntu uses ACPI for the devices you install it on.

on x86-64 yes. on some arm systems (mostly server ones) also yes. but we use Device-Tree on ARM laptops instead of ACPI. Device-Tree is defined in kernel source unlike ACPI which is defined in firmware. Thus, replacing firmware on ARM laptops usually doesn't give you better support in kernel (as Device-Tree bits are not in firmware, but part of kernel / distributions). However, it could give you a more convenient way to boot kernel than depthcharge

As for me, I did implement depthcharge support in kernel, so custom firmware is not a priority for me at all, but I think it is something that would be good to have so people can use more distributions which don't want to support depthcharge

Dae-Hang-Sook, 2024-04-04:
> > That's weird then. Ubuntu uses ACPI for the devices you install it on.
> 
> on x86-64 yes. on some arm systems (mostly server ones) also yes. but we use Device-Tree on ARM laptops instead of ACPI. Device-Tree is defined in kernel source unlike ACPI which is defined in firmware. Thus, replacing firmware on ARM laptops usually doesn't give you better support in kernel (as Device-Tree bits are not in firmware, but part of kernel / distribution). However, it could give you a more convenient way to boot kernel than depthcharge
> 
> As for me, I did implement depthcharge support in kernel, so custom firmware is not a priority for me at all, but I think it is something that would be good to have so people can use more distributions which don't want to support depthcharge 

Custom firmware for these devices would be super helpful. I emailed you by the way, my email ends in naver.com

jenneron, 2024-04-20:
@hexdump0815 i sent the image to them so this can be closed

closed 2024-07-13 by hexdump0815