velvet-os/imagebuilder #206: chromium and vivaldi fail to launch

szalony9szymek, 2024-04-01:
Both chromium and vivaldi throw the following error at me when started, with no window appearing.
I tried the --disable-gpu flag but it did not help. Help would be appreciated!

```
linux@changeme:~/Downloads$ chromium
[4898:4898:0401/194704.069678:ERROR:gbm_wrapper.cc(72)] Failed to get fd for plane.: No such file or directory (2)
[4898:4898:0401/194704.070275:ERROR:gbm_wrapper.cc(255)] Failed to export buffer to dma_buf: No such file or directory (2)
[4898:4898:0401/194704.070400:ERROR:gbm_wrapper.cc(72)] Failed to get fd for plane.: No such file or directory (2)
[4898:4898:0401/194704.070488:ERROR:gbm_wrapper.cc(255)] Failed to export buffer to dma_buf: No such file or directory (2)
[4898:4898:0401/194704.070584:ERROR:gbm_wrapper.cc(72)] Failed to get fd for plane.: No such file or directory (2)
[4898:4898:0401/194704.070667:ERROR:gbm_wrapper.cc(255)] Failed to export buffer to dma_buf: No such file or directory (2)
[4898:4898:0401/194704.070759:ERROR:gbm_wrapper.cc(72)] Failed to get fd for plane.: No such file or directory (2)
[4898:4898:0401/194704.070838:ERROR:gbm_wrapper.cc(255)] Failed to export buffer to dma_buf: No such file or directory (2)
[4898:4898:0401/194704.070928:ERROR:gbm_wrapper.cc(72)] Failed to get fd for plane.: No such file or directory (2)
[4898:4898:0401/194704.071010:ERROR:gbm_wrapper.cc(255)] Failed to export buffer to dma_buf: No such file or directory (2)
[4898:4898:0401/194704.071101:ERROR:gbm_wrapper.cc(72)] Failed to get fd for plane.: No such file or directory (2)
[4898:4898:0401/194704.071179:ERROR:gbm_wrapper.cc(255)] Failed to export buffer to dma_buf: No such file or directory (2)
[4898:4898:0401/194704.071269:ERROR:gbm_wrapper.cc(72)] Failed to get fd for plane.: No such file or directory (2)
[4898:4898:0401/194704.071348:ERROR:gbm_wrapper.cc(255)] Failed to export buffer to dma_buf: No such file or directory (2)
[4898:4898:0401/194704.071437:ERROR:gbm_wrapper.cc(72)] Failed to get fd for plane.: No such file or directory (2)
[4898:4898:0401/194704.071513:ERROR:gbm_wrapper.cc(255)] Failed to export buffer to dma_buf: No such file or directory (2)
[4898:4898:0401/194704.072965:ERROR:gbm_wrapper.cc(72)] Failed to get fd for plane.: No such file or directory (2)
[4898:4898:0401/194704.073070:ERROR:gbm_wrapper.cc(255)] Failed to export buffer to dma_buf: No such file or directory (2)
[4898:4898:0401/194704.073163:ERROR:gbm_wrapper.cc(72)] Failed to get fd for plane.: No such file or directory (2)
[4898:4898:0401/194704.073352:ERROR:gbm_wrapper.cc(255)] Failed to export buffer to dma_buf: No such file or directory (2)
[4898:4898:0401/194704.073447:ERROR:gbm_wrapper.cc(72)] Failed to get fd for plane.: No such file or directory (2)
[4898:4898:0401/194704.073527:ERROR:gbm_wrapper.cc(255)] Failed to export buffer to dma_buf: No such file or directory (2)
[4898:4898:0401/194704.073616:ERROR:gbm_wrapper.cc(72)] Failed to get fd for plane.: No such file or directory (2)
[4898:4898:0401/194704.073697:ERROR:gbm_wrapper.cc(255)] Failed to export buffer to dma_buf: No such file or directory (2)
[4898:4898:0401/194704.073790:ERROR:gbm_wrapper.cc(72)] Failed to get fd for plane.: No such file or directory (2)
[4898:4898:0401/194704.073868:ERROR:gbm_wrapper.cc(255)] Failed to export buffer to dma_buf: No such file or directory (2)
[4898:4898:0401/194704.073955:ERROR:gbm_wrapper.cc(72)] Failed to get fd for plane.: No such file or directory (2)
[4898:4898:0401/194704.074032:ERROR:gbm_wrapper.cc(255)] Failed to export buffer to dma_buf: No such file or directory (2)
[4898:4898:0401/194704.074119:ERROR:gbm_wrapper.cc(72)] Failed to get fd for plane.: No such file or directory (2)
[4898:4898:0401/194704.074196:ERROR:gbm_wrapper.cc(255)] Failed to export buffer to dma_buf: No such file or directory (2)
[4898:4898:0401/194704.074284:ERROR:gbm_wrapper.cc(72)] Failed to get fd for plane.: No such file or directory (2)
[4898:4898:0401/194704.074362:ERROR:gbm_wrapper.cc(255)] Failed to export buffer to dma_buf: No such file or directory (2)
```

hexdump0815, 2024-04-02:
can you please try running it from a commandline after setting "export LIBGL_ALWAYS_SOFTWARE=1" ?

szalony9szymek, 2024-04-02:
Seems to work now, thanks. Continues to work even after unsetting the flag, although with it it performs significantly better.

closed 2024-04-02 by szalony9szymek