velvet-os/imagebuilder #212: Mint?

tech-with-mo, 2024-05-29:
Is there a way to install Mint instead of Ubuntu because I couldn't find anything.

LukIsHere, 2024-05-29:
well in theory if u want to do it the hacky way
i once added kali repo to my source files and my computer started thinking it's kali instead of debian :3

so in theory if u replace ubuntu repos with mint repos and do sudo apt desktop upgrade it should change to mint (it's very likely mint version should be higher than ubuntu)

yet u will probably have to install cinamon (mint desktop environment iirc) manually
```
sudo apt install task-cinnamon-desktop
```
should do the trick

but yeah it's a very hacky (but simple) way that might cause so issues

## but

if all u want is the way desktop looks just install cinamon with command above ^ and set a wallpaper, it shouldn't really matter that much if u have ubuntu or mint repos in the background

archisman-panigrahi, 2024-09-22:
Mint developers haven't published any ARM images so far. You could manually add their repositories, but they only have packages for i386 and amd64.
Mint on Arm chromebooks are not happening anytime soon.

thenameisluk, 2024-09-23:
i have to admit i didn't know that

sorry for misinformation

'm not really sure what mint has to offer that ubuntu doesn't have that makes ppl **that use these images (which aren't end user read)** want it

archisman-panigrahi, 2025-09-03:
We can close this issue as it Mint does not offer ARM packages, and one can already install cinnamon in Ubuntu/Debian