velvet-os/imagebuilder #213: ℹ️ [GUIDE]: Getting Minecraft Java

LukIsHere, 2024-05-31:
# Version 1.16.5
_Note. this only works (well) on devices with gpu support/drivers_
1. Compile install [gl4es](https://github.com/ptitSeb/gl4es)
```git clone https://github.com/ptitSeb/gl4es.git```
```cd gl4es```
```mkdir build```
```cd build```
```cmake .. -DODROID=1 -DCMAKE_BUILD_TYPE=RelWithDebInfo```
```make -j8```
```sudo make install```
_Note. U can also use ```/scripts/install-gl4es.sh``` provided with the image but it doesn't seam to work currently_
_Sidenote. multiline codeblocks break number lists on github, so i used single line ones ^_
2. Export gl4es libGL.so
```export LD_LIBRARY_PATH=/usr/lib/gl4es/```
3. Install minecraft arm64
Install arm64 JRE from apt or anywhere else. Personally hallo1 recommends using the [dragonwell JDK 67](https://dragonwell-jdk.io/) for a free 20% performance boost and 30% less memory usage.
Install minecraft. You can use any launcher that supports ARM64 version of Minecraft. I am using [HMCL 71](https://github.com/huanghongxun/HMCL) that is confirmed working.
4. Install mods (optional)
The vanilla Minecraft works but performance is very poor(~30fps unstable)
Use the 1.16.5 version and install these mods:
- fabric
- fabric api 0.42.0
- sodium 0.2.0
- reese’s sodium options 1.4.7
_Note. these versions is confirmed working but you can use others as well._
**Important: Change some settings**
Launch your game and turn off the “Use compact vertex format” option in the video settings, or the game will crash when entering any world.
5. Enjoy Minecraft!
[image]
_Confirmed to work on [chromebook lenovo 10e](https://github.com/hexdump0815/imagebuilder/blob/main/systems/chromebook_kukui/readme.md) (debian trixie)_
**Credits for this part : [hallo1](https://forum.radxa.com/t/guide-run-minecraft-at-full-speed-on-rock-5b/11937)**
# Latest Version (1.20.6)
_Note: this only work on devices with opengl 3.3+ support (so no mali gpu for now)_
1. Install Java 21
- ```sudo apt install openjdk-21-jdk``` (assuming u r on debian trixie or sid)
or
- get [dragonwell JDK 67](https://dragonwell-jdk.io/)
2. Install launcher
download and install .deb file from link on 4th line of [this file](https://github.com/Botspot/pi-apps/blob/master/apps/Minecraft%20Java%20GDLauncher/install-64)
similar to this one ```https://github.com/Pi-Apps-Coders/files/releases/download/large-files/GDLauncher-linux-arm64-1.1.30-setup.deb```
_Note. why? in case there is a new version_
This version of launcher has special [meta repo](https://github.com/theofficialgman/piston-meta-arm64?tab=readme-ov-file) needed to work on arm
3. Set the Java path inside launcher settings
[image]
put path to ur java 21 in java 17 section (if u used apt method it should be ```/usr/lib/jvm/java-21-openjdk-arm64/bin/java```)
_Note. why? because newer versions of minecraft run on java 21_
4. Enjoy Minecraft!
[image]
_Confirmed to work on chromebook [lenovo IdeaPad Duet 5](https://github.com/hexdump0815/imagebuilder/blob/main/systems/chromebook_trogdor/readme.md) (debian trixie)_

_Note. in theory we can use software renderer to run versions of mc on any system but the performance would be **very** poor_

_Sidenote. in theory it is possible to run minecraft bedrock edition on devices with gpu support/drivers (opengl 3.1 and even lower so yes mali) with [this project](https://github.com/minecraft-linux/mcpelauncher-manifest) but the [comunity was unable to help me with that](https://github.com/minecraft-linux/mcpelauncher-manifest/issues/983). So ig that's for me to figure out_

_PS. for the powerful gaming devices which chromebooks definitely are :3 have a fun time playing_
~luk

linsyking, 2024-07-14:
Why using `gl4es` instead of using the default mesa gl library?

LukIsHere, 2024-07-14:
@linsyking for [adreno gpu's](https://docs.mesa3d.org/drivers/freedreno.html) it's not required (and can be skipped) since these r very well supported
but for [mali gpu's](https://docs.mesa3d.org/drivers/panfrost.html) it won't work without gl2es