velvet-os/imagebuilder #214: f2fs for rootfs

nonameCCC, 2024-06-01:
Is there any possibly to use f2fs as the root filesystem?
I tried to do debootstrap and kernel installation manually on a Chromebook trogdor,but it seems that the kernel can't use f2fs as the root filesystem.

hexdump0815, 2024-06-01:
i think i added f2fs support after i built the kernels for the latest images maybe - https://github.com/hexdump0815/kernel-config-options/commit/bf08a134114f12b5b7bbbefedd5fba8c527dae76 ... what might also be a problem is that it is defined as =m and not as =y in the kernel config ... maybe you should build you own kernel with it set to =y ... please let me know if this works

general note: i do not plan to add support for f2fs as rootfs to the framework to keep its complexity somehow managable, so using f2fs as rootfs will always be a bit roll-your-own ...

nonameCCC, 2024-06-02:
> i think i added f2fs support after i built the kernels for the latest images maybe - [hexdump0815/kernel-config-options@bf08a13](https://github.com/hexdump0815/kernel-config-options/commit/bf08a134114f12b5b7bbbefedd5fba8c527dae76) ... what might also be a problem is that it is defined as =m and not as =y in the kernel config ... maybe you should build you own kernel with it set to =y ... please let me know if this works
> 
> general note: i do not plan to add support for f2fs as rootfs to the framework to keep its complexity somehow managable, so using f2fs as rootfs will always be a bit roll-your-own ...

Thanks for your advice! I may try that out later.