velvet-os/imagebuilder #215: Specify a keymap for Luks password during boot ?

gallee, 2024-06-01:
Hello,

I managed to install on my kukui jammy with a LUKS full disk encryption :heart_eyes: 

However, I thought I missed something because my keymap is specific (fr_FR), and I was getting `no key available with this passphrase****` :sweat: 
Now that I know I should use the US keymap, I manage to boot but I would like to specify my keymap. I searched but did not find much (like this [post](https://forum.studentb.eu/index.php/ubuntu/how-to-set-keyboard-layout-for-a-luks-encrypted-disk-on-ubuntu-2204/))

Is it possible to configure the keymap when typing the Luks passwork at boot time ?

Thanks for your work !

hexdump0815, 2024-06-01:
congratulations for the luks install ... i'm not fully sure about this, but do you have the proper keyboard defined in /etc/default/keyboard - it might be that update-iniramfs takes and defines it from there, but as said i'm not sure ...

gallee, 2024-06-22:
I managed to have the initamfs configured correctly :tada: . I think this is because my /etc/default/keyboard was not set correctly as you suggested.

closed 2024-06-22 by gallee