velvet-os/imagebuilder #216: [kukui][juniper] Unable to boot with 6.6.9-stb-mt8+ and luks

gallee, 2024-06-01:
Hello,

I want to update my luks on emms ubuntu jammy install currently running  `6.1.51-stb-mt8+` with the last kernel available for download i.e. [6.6.9-stb-mt8+] (https://github.com/hexdump0815/linux-mainline-mediatek-mt81xx-kernel/releases/tag/6.6.9-stb-mt8%2B)

I applied the script mentionned on the [documention](https://github.com/hexdump0815/imagebuilder/blob/main/doc/install-to-emmc-with-luks-full-disk-encryption.txt)

However, when I boot it, I have a debian 12 image display (contrary to the 6.1 kernel where it is an xubuntu) and I have the following:

```
Gave up waiting for root file system device. Common problems:
Boot args (cat /proc/cmdline) Check rootdelay= (did the system wait long enough?)
Missing modules (cat /proc/modules: ls /dev)
ALERT! LABEL=rootemmc does not exist. Dropping to a shell!
BusyBox v1.35.0 (Debian 1:1.35.0-4+b3) built-in shell (ash) Enter 'help' for a list of built-in commands.
(initramfs)
```

Is this kernel only for debian 12 or am I missing something ?

hexdump0815, 2024-06-01:
did you run the update-initramfs from https://github.com/hexdump0815/imagebuilder/blob/main/doc/install-to-emmc-with-luks-full-disk-encryption.txt#L225-L227 and this https://github.com/hexdump0815/imagebuilder/blob/main/doc/install-to-emmc-with-luks-full-disk-encryption.txt#L251-L283 ?

gallee, 2024-06-01:
Yes I did

gallee, 2024-08-17:
I do not know what I missed : I re apply the procedure and it was ok !

closed 2024-08-17 by gallee