velvet-os/imagebuilder #219: Problems while constructing own system

nonameCCC, 2024-06-03:
The image file works fine on my wormdingler, but I tried to construct my own system with debootstrap as I want to use GNOME and gdm for better touchscreen experience, and f2fs for better performance on emmc.

But there was several problems:

1. Self-built kernel with f2fs kernel options and apparmor cmdline: can't boot up,with lots of FAILED on systemd
2. Pre-built cbq+ kernel with btrfs: couldn't detect network hardware,even with proper extra files cloned from GitHub 

And I find out that in situation 1, if no root password and new users were set,the system could boot into a broken GNOME environment with the same network issue of situation 2.

I know that building system without completely following the guide was not recommended, but those problems were really puzzling.

LukIsHere, 2024-06-03:
not sure why u do this to urself
but i think this is what u r looking for https://github.com/hexdump0815/qcom-tools-build ?

hexdump0815, 2024-06-05:
@nonameCCC - if your goal is only to get gnome: it should be possible to simply install it via apt on the regular xfce image and then switch to it ... i think quite a few people already did that with both gnome and kde

LukIsHere, 2024-06-05:
maybe should add instruction on installing different desktop environments in docs
simple mentioning task-gnome-desktop, task-kde-desktop ...?