velvet-os/imagebuilder #221: Can't boot into debian or Ubuntu aarch64 image, beeps when I press Ctrl+U

jonathan011Hh, 2024-06-06:
Can't boot into debian or Ubuntu aarch64 image, beeps when I press Ctrl+U, (I followed all the instructions and I made sure my device was supported I'm currently using a Lenovo IdeaPad 3 mt8183 chromebook
architecture is "aarch64" 
thx for ur time all help  is appreciated

LukIsHere, 2024-06-06:
hi
u actually need to enable developer mode and booting from usb for this to work

Turn off the chromebook, then press and hold volume up, volume down, and power simultaneously until menu pops up (around 10 or so seconds).
In there enable developer mode.

After that go back to chromeos, it should be reseted, don't set it up again. Press ctrl+alt+Refresh, login as chronos (no password).
Type ```sudo crossystem dev_boot_usb=1 dev_boot_signed_only=0```
You can also set gbb flags there, look [here](https://wiki.postmarketos.org/wiki/Category:ChromeOS#Installation)

After rebooting you should be able to boot from external device using menu.
Also more details on process i described should be [here](https://wiki.postmarketos.org/wiki/Category:ChromeOS#Installation)

LukIsHere, 2024-06-06:
also isn't ideapad duet 3 	Qualcomm Snapdragon 7c?
https://wiki.postmarketos.org/wiki/Lenovo_IdeaPad_Duet_3_(google-wormdingler)

jonathan011Hh, 2024-06-06:
hey, I did all of those things it just, beeps after hang of  black screen and beeps lol don't know what to do my device is in dev mode and also supported:(

jonathan011Hh, 2024-06-06:
not yet, I'll try that, :/

LukIsHere, 2024-06-06:
ok lemme know if it works👍

jonathan011Hh, 2024-06-06:
one question were is exactly the "7c" image I cant seem to find it?

LukIsHere, 2024-06-06:
'll tell u just quick question
is this your device https://pcsupport.lenovo.com/us/en/products/chromebook-laptops/lenovo-chromebooks-series/ideapad-3-chrome-14m836/documentation/doc_userguide
or this one https://pcsupport.lenovo.com/us/pl/products/chromebook-laptops/lenovo-chromebooks-series/duet-3-chrome-11q727/document-userguide/doc_userguide ?

jonathan011Hh, 2024-06-06:
I found it its downloading, how do I boot from it I pressed Ctrl+U it beeps(earlier) so hopefully when I do this it'll boot properly if not I might have to go a different route I'll post updates.

LukIsHere, 2024-06-06:
if the first one the 7c image won't work
just send a video if u can
and also tell me how you flashed the usb or sd card?
it usually does the beep if it doesn't detect a device to boot from

jonathan011Hh, 2024-06-06:
oh ok, also my device is the top link u sent me, asking which device I have(Lenovo IdeaPad 3 Chromebook)
updates coming soon

jonathan011Hh, 2024-06-06:
it did not work im unable to send a video rn, is there anything else i can do its at dev screen i press keys to boot and it goes black and then back to the screen and beeps very loudy lol idkw to do:<

jonathan011Hh, 2024-06-06:
> if the first one the 7c image won't work just send a video if u can and also tell me how you flashed the usb or sd card? it usually does the beep if it doesn't detect a device to boot from

well I did enable dev USB booting etc, and I booted using control+U keys.

LukIsHere, 2024-06-06:
also are you sure you have arm variant? there is also x64 variant
https://pcsupport.lenovo.com/us/en/products/chromebook-laptops/lenovo-chromebooks-series/ideapad-3-cb-14igl05/documentation/doc_userguide

jonathan011Hh, 2024-06-06:
"14M836" hopefully that's the M.O.

jonathan011Hh, 2024-06-06:
model number*

jonathan011Hh, 2024-06-06:
yes I don't have Intel or AMD or armv9l or anything like that its a (CPU) "mediatek Kamponio 500 mt8183"
architecture "Aarch64"

LukIsHere, 2024-06-06:
ok i checked it is an arm one
https://psref.lenovo.com/syspool/Sys/PDF/IdeaPad/IdeaPad_3_CB_14M836/IdeaPad_3_CB_14M836_Spec.pdf
will have a look into hardware maintenance manual for more info gimme a moment

jonathan011Hh, 2024-06-06:
ok

LukIsHere, 2024-06-06:
did you use usb or micro sd card?
how did you flash it?

jonathan011Hh, 2024-06-06:
i used usb, and i put the file u told me to download into the usb slot and then i rebooted and pressed ctrl+U

jonathan011Hh, 2024-06-06:
I did use it I just enabled it again to make sure

LukIsHere, 2024-06-06:
have you flashed the file or just put it onto the usb?

jonathan011Hh, 2024-06-06:
just put it into the USB.

LukIsHere, 2024-06-06:
bruh

jonathan011Hh, 2024-06-06:
oh how do I flash it(srry)

LukIsHere, 2024-06-06:
you need to flash the file
what system do u have on your main pc?

jonathan011Hh, 2024-06-06:
Mac and windows

jonathan011Hh, 2024-06-06:
like using belena etcher? and then boot it onto chromebook

jonathan011Hh, 2024-06-06:
*for example(belenaetcher)*

LukIsHere, 2024-06-06:
on on windows use this https://sourceforge.net/projects/win32diskimager/
select the mt8183 image
select your usb
click write

LukIsHere, 2024-06-06:
after that just plug it in and it should boot

jonathan011Hh, 2024-06-06:
ok I'll give it a shot, let u know if its succesfull

jonathan011Hh, 2024-06-06:
can I use etcher tho? my windows takes for ever to boot up (I have windows 11 loaded on MacBook pro).

LukIsHere, 2024-06-06:
balena etcher works too

jonathan011Hh, 2024-06-06:
thx giving it a shot.

jonathan011Hh, 2024-06-06:
I tried it it still beeps

jonathan011Hh, 2024-06-06:
doesn't boot whatsoever

LukIsHere, 2024-06-06:
i forgot the image is compressed
try unpacking it with some archive program so it's .img and then flashing

jonathan011Hh, 2024-06-06:
oh ok did that testing it now

jonathan011Hh, 2024-06-06:
it dosnt work still😔

jonathan011Hh, 2024-06-06:
my USB and USB port r fine it just doesn't want to boot lol

LukIsHere, 2024-06-06:
weird
you know mac os had dd so you can try using that

on ur mac
first ```diskutil list``` fin ur usb
then ```dd of=image.img if=/dev/yourUSB status=progress```

jonathan011Hh, 2024-06-06:
oh well I cant install it on my chromebook

jonathan011Hh, 2024-06-06:
?

LukIsHere, 2024-06-06:
you can kinda install it directly but it is very risky

jonathan011Hh, 2024-06-06:
no I mean just boot it off of a USB, also is there a way to get legacy on Aarch64 architecture

LukIsHere, 2024-06-06:
not sure what you mean by legacy boot

jonathan011Hh, 2024-06-06:
nvm I mean like is there a way to boot from USB on a arm Chromebook idkw but it still isn't working:/

LukIsHere, 2024-06-06:
try setting the gbb flags if u haven't it might help make sure you have usb boot enabled

LukIsHere, 2024-06-06:
read the gbb flags section
https://wiki.postmarketos.org/wiki/Category:ChromeOS

jonathan011Hh, 2024-06-06:
I cant disable write protection it says "failed to apply new settings: unknown WP error"

LukIsHere, 2024-06-06:
yeah don't worry about that just ```futility gbb --set --flash --flags=0x19``` it should work anyway

jonathan011Hh, 2024-06-06:
oh I got error=  that I have to disable WP(write-protection)

LukIsHere, 2024-06-06:
weird your device shouldn't have hardware one
could you pls try running
```sudo crossystem dev_boot_usb=1 dev_boot_signed_only=0```
again and see if it solves the problem
if it doesn't it's probably usb flashed incorrectly

jonathan011Hh, 2024-06-06:
I already did lol like 6 times it doesn't like it and I did use UNETBOOTIN to write to USB if that changes anything

LukIsHere, 2024-06-06:
it should be one of this images https://github.com/hexdump0815/imagebuilder/blob/main/systems/chromebook_kukui/readme.md
your probably want exactly this one https://github.com/hexdump0815/imagebuilder/releases/tag/230917-01
**uncompressed** with balena etcher

jonathan011Hh, 2024-06-06:
ok Im trying different file and I already tried thos

jonathan011Hh, 2024-06-06:
ok trieing it lets c ifdis works

jonathan011Hh, 2024-06-06:
it still beeps:<  😭

LukIsHere, 2024-06-06:
damn 'm out of ideas
can you try with micro sd card?
provide a recording?

[4 more replies]

closed 2024-06-06 by jonathan011Hh