velvet-os/imagebuilder #222: Hi there I'm unable to boot into live image I have dev mode on and USB booting and it simply wont boot.

jonathan011Hh, 2024-06-07:
Hi there I'm unable to boot into live image I have dev mode on and USB booting and it simply wont boot. I know I've already posted to the issues page but I'm wondering if someone can help me Ive spent forever trying to get this working I have a "Lenovo IdeaPad 3 Chromebook" with MediaTek Kompanio 500, MT8183"
all help is appreciated (yes my device is supported)

LukIsHere, 2024-06-07:
question
does your developer mode menu look like this
[image]
or like this
[image]
?

LukIsHere, 2024-06-07:
if the first one try going to developer options and selecting boot from usb/external device
if the second one, (assuming usb boot is enabled and usb device/port works) u probably still have flashed the image incorrectly

jonathan011Hh, 2024-06-07:
oh ok

jonathan011Hh, 2024-06-07:
I'll try flashing the image

jonathan011Hh, 2024-06-07:
no my dev screen looks like the bottom one

jonathan011Hh, 2024-06-07:
I flashed it, it said flash succesfull on belena etcher? it still just beeps

jonathan011Hh, 2024-06-07:
[image]
here screenshot is this the right file to download as iso or os

LukIsHere, 2024-06-08:
yes the file is correct, the bulleye is pretty old one thoe
i would recommend you going with bookworm : https://github.com/hexdump0815/imagebuilder/releases/tag/230917-01
have you tried using [raspberry pi imager](https://www.raspberrypi.com/software/)?
[image]
seams to be producing bootable usb for me (tested it with my kukui device), is cross platform
and you can also pass compressed version (.img.gz) to it

jonathan011Hh, 2024-06-08:
I have right file and did everything right it just wont boot( is it because I have a blue 3.0 USB)

jonathan011Hh, 2024-06-08:
yes I will try pi imager

jonathan011Hh, 2024-06-08:
it still turns to black screen then hangs there for like 10 seconds then beeps loudly

hexdump0815, 2024-06-08:
do you press ctrl-u when you see the dev screen to force it to boot from usb? otherwise it will wait a while, then beep and then boot chromeos ... if that is not the problem then i'm not sure what the problem is - it should at least boot ... maybe you can check your flashed usb device on some other system and check that it has four partitions ("fdisk -l" or "cat /proc/partitions" on linux as root for instance) - if not then something went wrong with the flashing i guess

jonathan011Hh, 2024-06-08:
idk I tried all of that

jonathan011Hh, 2024-06-08:
lmk check tho

jonathan011Hh, 2024-06-08:
ya it has more than 4

jonathan011Hh, 2024-06-08:
here's my USB=  
[image]
USB 3.0 64GB

LukIsHere, 2024-06-08:
it might be one of these fake capacity usb's?
honestly no other idea what might be going wrong

jonathan011Hh, 2024-06-08:
? oh i have a 64GB USB 3.0, usb might be bad or there isnt an image to boot from or it dosnt like it for whatever reason.

LukIsHere, 2024-06-08:
it's like when usb says it has x capacity but it actually has y
maybe try with different usb?

jonathan011Hh, 2024-06-08:
ok will do, (I tried a SD card still didn't boot) also it didn't boot it like went black like it was trying to boot then quit.

LukIsHere, 2024-06-08:
ok make sure you use kukui image

jonathan011Hh, 2024-06-08:
aight ima test it

jonathan011Hh, 2024-06-08:
it keeps beeping also, take a look 
[Screen recording 2024-06-08 6.01.34 PM.webm](https://github.com/hexdump0815/imagebuilder/assets/162735293/31856eeb-450c-491a-a8e8-2cb05a567db6)
idkw its saying this stuff, also I'm trying this alternatively also I did flash it earlier!

LukIsHere, 2024-06-08:
but that's not how u flash an image

LukIsHere, 2024-06-08:
if you want to flash the image from your chromebook
open chromebook terminal
download the image
```
wget https://github.com/hexdump0815/imagebuilder/releases/download/230917-01/chromebook_kukui-aarch64-bookworm.img.gz
```
unpack the image
```
gunzip chromebook_kukui-aarch64-bookworm.img.gz
```
find your usb device
```
lsblk
```
_output:_
```
luk@chluk /mnt $ lsblk
NAME         MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
mtdblock0     31:0    0     8M  0 disk 
sda1      179:0    0 116,5G  0 disk <-- you are looking for entry with the size of your usb
├─sda1p1  179:1    0    32M  0 part <-- not this, this is a partition
├─sda1p2  179:2    0    32M  0 part 
├─sda1p3  179:3    0   512M  0 part /boot
├─sda1p4  179:4    0 108,5G  0 part /
└─sda1p5  179:5    0   7,5G  0 part [SWAP]
```
flash the image
```
sudo dd status=progress if=chromebook_kukui-aarch64-bookworm.img of=/dev/sda1 <-- replace sda1 with you usb name from the step above
```
device should be ready to go
if there is any problem with a command just add sudo before it

jonathan011Hh, 2024-06-08:
gzip: chromebook_kukui-aarch64-bookworm.img.gz: No such file or directory

LukIsHere, 2024-06-08:
u haven't downloaded the image using the command or the download failed

jonathan011Hh, 2024-06-08:
I don't have wget

LukIsHere, 2024-06-08:
how about 
```
curl https://github.com/hexdump0815/imagebuilder/releases/download/230917-01/chromebook_kukui-aarch64-bookworm.img.gz
```?

jonathan011Hh, 2024-06-08:
it worked ima extract image

LukIsHere, 2024-06-08:
wait curl is more complicated

LukIsHere, 2024-06-08:
it should actully be 
```
curl -O -L https://github.com/hexdump0815/imagebuilder/releases/download/230917-01/chromebook_kukui-aarch64-bookworm.img.gz
```

jonathan011Hh, 2024-06-08:
ok it also says no space left stay on issue I'm having dinner ttyl I'll tell u if this works!

jonathan011Hh, 2024-06-08:
curl: (23) Failure writing output to destination, passed 16366 returned 138

LukIsHere, 2024-06-08:
try with sudo

jonathan011Hh, 2024-06-08:
ok

jonathan011Hh, 2024-06-08:
still no

jonathan011Hh, 2024-06-08:
gzip: chromebook_kukui-aarch64-bookworm.img.gz: unexpected end of file

LukIsHere, 2024-06-08:
```
curl -o chromebook_kukui-aarch64-bookworm.img.gz -L https://github.com/hexdump0815/imagebuilder/releases/download/230917-01/chromebook_kukui-aarch64-bookworm.img.gz
```
?

jonathan011Hh, 2024-06-08:
still no

LukIsHere, 2024-06-08:
```
find / -name chromebook_kukui-aarch64-bookworm.img.gz 2>/dev/null
```
share the output

jonathan011Hh, 2024-06-08:
I found it its running the gzip script rn

jonathan011Hh, 2024-06-08:
hopefully it works its flashing it rn

jonathan011Hh, 2024-06-09:
it worked!😁(its slow to boot tho lol) question am I able to use this for other operating systems? if so how?