velvet-os/imagebuilder #223: Anyway to hardware video

FrostNovaHD, 2024-06-08:
I can't seem to get it working on trogdor

hexdump0815, 2024-06-08:
it should be possible on trogdor - i did not look deeper into it ... maybe having a look at the the postmarketos trogdor related pmaports dirs might be a good idea as i think they have working what is possible on trogdor ... in case you find out anything or anything which is missing, please post a quick update here, so that i can potentially update it for future images

WesleyVanNeck, 2024-06-23:
> it should be possible on trogdor - i did not look deeper into it ... maybe having a look at the the postmarketos trogdor related pmaports dirs might be a good idea as i think they have working what [...]

Hi hexdump I just found that gstreamer stuff was missing and mesa va drivers

LukIsHere, 2024-06-25:
i looked into it
there is that thing called [venus](https://wiki.postmarketos.org/wiki/Hardware_video_acceleration) it has v4l2m2m (Video for Linux 2 Memory-to-Memory) api
and ffmpeg should support it maybe
can look into it further when back home

LukIsHere, 2024-06-25:
yeah ffmpeg doesn't seam to work
it either seqfaults or just hangs for no reason
```
ffmpeg -i out2.mp4 -pix_fmt nv12 -c:v h264_v4l2m2m out3.mp4
...
[h264_v4l2m2m @ 0xaaab147de710] Using device /dev/video1
[h264_v4l2m2m @ 0xaaab147de710] driver 'qcom-venus' on card 'Qualcomm Venus video encoder' in mplane mode
[h264_v4l2m2m @ 0xaaab147de710] requesting formats: output=NV12/nv12 capture=H264/none
Output #0, mp4, to 'out3.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    encoder         : Lavf60.16.100
  Stream #0:0(und): Video: h264 (avc1 / 0x31637661), nv12(tv, bt709, progressive), 1914x1002 [SAR 1:1 DAR 319:167], q=2-31, 200 kb/s, 9.92 fps, 15232 tbn (default)
    Metadata:
      handler_name    : VideoHandler
      vendor_id       : [0][0][0][0]
      encoder         : Lavc60.31.102 h264_v4l2m2m
  Stream #0:1(und): Audio: aac (LC) (mp4a / 0x6134706D), 44100 Hz, mono, fltp, 69 kb/s (default)
    Metadata:
      handler_name    : SoundHandler
      vendor_id       : [0][0][0][0]
      encoder         : Lavc60.31.102 aac
and nothing
```
i checked the kernel config and venus is enables so idk what is worng

linsyking, 2024-07-07:
Hardware decoder is working on my device.

Encoder doesn't work.

I used mpv:

```
mpv --hwdec=v4l2m2m-copy video.mp4
```

and it worked, but it's slower than software decoder. Not sure why.

linsyking, 2024-07-08:
I could use `ffplay` to decode video:

```bash
ffplay -codec:v h264_v4l2m2m a.mp4
```

But I cannot make `mpv` working, it always complains the `hwdec` not supported.

The encoder doesn't work.

Using `h264_v4l2m2m` uses more CPU (140%) while the software decoder uses only 30%. I'm not sure why.