velvet-os/imagebuilder #224: hi I'm booting into the bookworm iso or image and its currently booting fine but it takes forever and keeps saying something about disabling things and "mt8183 sound mic controller" smth like that

jonathan011Hh, 2024-06-09:
hi I'm booting into the bookworm iso or image and its currently booting fine but it takes forever and keeps saying something about disabling things and "mt8183 sound mic controller" smth like that, any help. (Lenovo IdeaPad 3 MT8183) {non-intel}
all help is appreciated

LukIsHere, 2024-06-09:
mt8183 is an arm64 processor so there is really no need to say "no intel"
when booting from a usb u r limited by ur usb speed
you can install the system directly to memory and it would be way faster

hexdump0815, 2024-06-09:
@jonathan011Hh - can you please stop always creating new github issues for essentially the same topic - lets please try to continue within one github issue with your journel to get your chromebook working

in general the arm chromebooks are not as well supported as intel system, so having some things not working is kind of expected and getting some of the non working things working might not be possible at the moment (as support in the linux kernel for instance is still missing) or would require own research as nobody took care about it yet. useful input for getting forward is searching through the existing github issues here as some of the topics have already been brought up and/or discussed in other issues and the readme in chromebok_kukui where i tried to write down some basic known things.

jonathan011Hh, 2024-06-09:
ok I'm srry I'll stick to researching my own terms of issues, as all ways thx for helping me so much I actually figured this out and got it to boot(sort of) after a week of trying, thx alot ima go play around some more with it thx alot [hexdump0815](https://github.com/hexdump0815) and Luke.