velvet-os/imagebuilder #225: How to create a Chromebook bootable image of Raspberry Pi OS?

archisman-panigrahi, 2024-06-11:
The Raspberry Pi OS is extremely lightweight, but it is tweaked to have many useful features.

I tried flashing Raspberry Pi OS using the RPi Imager app to a USB drive, but it does not boot when I press `ctrl + U` in my Chromebook. However, the debian image (from this repo) or postmarketOS just boots.

Perhaps the RPi OS image needs to be configured in a way so that the Chromebook can boot it.

Can someone give me some pointers on how I should start?

I am using a Lenovo 300e (HANA).

LukIsHere, 2024-06-12:
chromebooks just work different
you can't just boot random images on a chromebook funny :3
Raspbian is made for RPI not chromebooks
there aren't any tweaks you can benefit from (afaik)
not sure what are you trying to achieve
- have raspbian repos
Just add it
```
# add the raspberry pi os gpg key
wget -O - http://archive.raspberrypi.org/debian/raspberrypi.gpg.key | sudo apt-key add -
# add the raspberry pi os apt repository
echo deb http://archive.raspberrypi.org/debian/ $(lsb_release -cs) main | sudo tee /etc/apt/sources.list.d/raspios.list
# update repositories
sudo apt update
```
- have ur desktop look like on raspberry pi
Add repo and run
```
# install pixel
sudo apt install raspberrypi-ui-mods -y
```
[more details](https://i12bretro.github.io/tutorials/0738.html)

archisman-panigrahi, 2024-06-13:
Thank you. 

BTW, I just found that there is a patch for the drivers for Chromebook HANA https://www.phoronix.com/news/PowerVR-GX6250-MT8173. Hopefully it will make the web browsing experience in Debian as smooth as in Chrome OS.

closed 2024-06-13 by archisman-panigrahi

hexdump0815, 2024-06-14:
just as a note: this gpu patch is still in very early stages and it will still take quite a while until it will be useable for anything useful.

archisman-panigrahi, 2024-09-24:
>  there aren't any tweaks you can benefit from (afaik)

The desktop environment is very lightweight, yet it looks quite good and is more convenient to use compared to vanilla XFCE - of course XFCE can be customized. That is all.