velvet-os/imagebuilder #226: MT8183/ Chromebook kukui New GPU Patches Available

PD-Kerman, 2024-06-14:
Hi there, I've am using the Debian image under this project on my kukui device and it's really good thank you for your work

But I noticed that I've got some unusual stutter during some daily tasks so I went and ran the glmark, and I noticed that it only got 300pts or so when postmarketos got around 1000pts, and I found out that there was a new kernel patch for this to make the gpu much faster than before here:
https://gitlab.com/postmarketOS/pmaports/-/tree/master/device/community/linux-postmarketos-mediatek-mt8183?ref_type=heads

So I'm wondering is there anyway for us to apply this kernel patch for the gpu?

And I'm hoping for some image upgrades since the latest version of kukui image was from last year

Thank u for your great work on the project which made my machine better than ever~
It would be even better if this could get fixed and we could get updates~
Thanks in advanced.

Best Regards
PD-Kemman

PD-Kerman, 2024-06-14:
Follow update:
I just applied the upgrade to the kernel and activated the kernel 6.6.9 and the gpu performance did get better, I ran the glmark2 it got around 500pts but still kinda far away from 1000pts which the exact same hardware got on pmOS 
:D

gallee, 2024-06-22:
Hello,

For what it's worth on `6.6.9` with jammy on my kukui / juniper, I have a glmark2 score of 739

hexdump0815, 2024-06-22:
the glmark result also depends on the window-system and -manager a bit - sometimes it can get bigger with xorg/xfce if "display compositing" is enabled in settings -> window manager tweaks -> compositing - sometimes also not or it can be even slower then ... wayland and xorg will also give different results usually ...

PD-Kerman, 2024-06-22:
> the glmark result also depends on the window-system and -manager a bit - sometimes it can get bigger with xorg/xfce if "display compositing" is enabled in settings -> window manager tweaks -> [...]

Thank you for your replies, well yeah the DE and the windows manager could be considerable variants, although I'm pretty sure I made those var as little as possible in the first place tho,
Because I personally consider showing contents when dragging/resizing windows is a waste of power/performance so I toggle the windows manager settings every time I did a fresh installation, especially in this case xfce and on a platform with a relatively low performance, so yeah I think they should be exactly the same when it comes to windows manager settings, compositor etc. and considering that I'm using xfce on both systems so xorg/wayland shouldn't be a considerable var then. So the GPU performance gap still should be at a considerable level, anyhow, even though there was some slight differences caused by DE/Compositor because there's almost a 100% gap between these two.

But, yeah it sure would be better if i perform a second test with specific those var checked, I think I could get that done within 12hrs.

Thank you for your replies again, hope you're having a great day
Sincerely With Best Regards,
PD-Kerman

PD-Kerman, 2024-06-23:
Okay I found a way to *port or simply compile and install the kernel from pmOS maybe on your release which I'm planning to try these days, and pardon me if I'm a little bit busy this weekend but I'll perform the test asap

archisman-panigrahi, 2024-09-22:
I read that the driver for this GPU was sent to the Kernel list, but I am unsure if it has been accepted https://www.phoronix.com/news/PowerVR-GX6250-MT8173

tech-with-mo, 2024-09-29:
Hi, is there a proper guide to applying kernel patches? Couldn't find any

thenameisluk, 2024-09-29:
@tech-with-mo 
https://github.com/hexdump0815/linux-mainline-mediatek-mt81xx-kernel/blob/master/readme.mt8
here are all steps for compiling kernel for kukui chromebooks with patches

there is no guide i am aware of on obtaining them