velvet-os/imagebuilder #228: chromebook_corsola: status: magneton (lenovo chromebook slim 3)

LukIsHere, 2024-07-02:
# Notes on Magneton running 6.7.0-rc8-cos-mt9+
[image]

Here is list of working and non working features on the device
### Working
- wifi
- bluetooth
- builtin display
- energy: battery status
- USB C and A for basic usb functionality
- gpu opengl: via mesa panfrost (opengl 3.1)
[image]
- external display (via usb-c to hdmi adapter)
[image]
my honest reaction
[image]
- webcams (to be expected on **laptops**)
[image]

### untested

- hw video encode/decode: should work via [mtk-vpu](https://wiki.postmarketos.org/wiki/Hardware_video_acceleration)

### broken
- sound: speaker (temporarily disabled by hexdump for reasons, bluetooth is a working option)
- suspend/resume (is kinda iffy)

### Installation

coming soon (before year of linux desktop :3)

### PROBLEMS

```
luk@changeme:/$ ls -lh
total 20K
lrwxrwxrwx   1 root root    7 Sep  3  2023 bin -> usr/bin
drwxr-xr-x   4 root root 4.0K Jan 12 19:10 boot
drwxr-xr-x  20 root root 5.0K Jul  2 22:42 dev
drwxr-xr-x   1 root root 3.5K Jul  2 22:48 etc
drwxr-xr-x   1 root root    6 Jul  2 22:23 home
lrwxrwxrwx   1 root root    7 Sep  3  2023 lib -> usr/lib
drwxr-xr-x   1 root root    0 Sep  3  2023 media
drwxr-xr-x   1 root root    0 Sep  3  2023 mnt
drwxr-xr-x   1 root root    0 Sep  3  2023 opt
dr-xr-xr-x 362 root root    0 Jul  2 22:20 proc
drwx------   1 root root  322 Jul  2 22:22 root
drwxr-xr-x  27 root root  720 Jul  2 22:42 run
lrwxrwxrwx   1 root root    8 Sep  3  2023 sbin -> usr/sbin
drwxr-xr-x   1 root root  872 Jan 12 19:07 scripts
drwxr-xr-x   1 root root    0 Sep  3  2023 srv
drwxr-xr-x   1 root root   12 Jan 12 19:17 swap
dr-xr-xr-x  13 root root    0 Jul  2 22:20 sys
drwxrwxrwt   1 root root 1.1K Jul  2 22:48 tmp
drwxr-xr-x   1 luk  luk    84 Jun 26  2023 usr
drwxr-xr-x   1 root root   90 Sep  3  2023 var
```
on latest image /usr directory seams to have wrong ownership
```
luk@changeme:/$ sudo /scripts/extend-rootfs.sh 
[sudo] password for luk: 

resizing root filesystem

cannot determine the root device base - giving up
```
also that ^, same issue when updating initramfs
### The end
overall the device works/looks great, the only thing that i miss is audio from speakers
_Sidenote. will use this device to do and understand luks emmc instalation and hopefully rewrite the guide on it_

hexdump0815, 2024-07-04:
oh - nice that the external monitor works ... there is a good chance that things will work even better with a more recent version from that kernel tree via which most of the mediatek chromebook changes come into the linux mainline kernel ... regarding that permission problem for /usr i'll have to check where it comes from (most probably it is related to my local image build env or how i built this image) - thanks a lot for spotting this

LukIsHere, 2024-07-04:
@hexdump0815 what's exactly with the sound? high pitch noise again or sth?

LukIsHere, 2024-07-04:
@hexdump0815 
on the ownership topic there is some more logs
```
Detected unsafe path transition / (owned by luk) → /run (owned by root) during canonicalization of /run.
Detected unsafe path transition / (owned by luk) → /run (owned by root) during canonicalization of /run.
Detected unsafe path transition / (owned by luk) → /etc (owned by root) during canonicalization of /etc.
Detected unsafe path transition / (owned by luk) → /var (owned by root) during canonicalization of /var.
Detected unsafe path transition / (owned by luk) → /var (owned by root) during canonicalization of /var.
Detected unsafe path transition / (owned by luk) → /var (owned by root) during canonicalization of /var/log.
Detected unsafe path transition / (owned by luk) → /run (owned by root) during canonicalization of /run/lock.
Detected unsafe path transition / (owned by luk) → /run (owned by root) during canonicalization of /run/systemd.
Detected unsafe path transition / (owned by luk) → /run (owned by root) during canonicalization of /run/systemd/netif.
Detected unsafe path transition / (owned by luk) → /run (owned by root) during canonicalization of /run/systemd/netif.
Detected unsafe path transition / (owned by luk) → /run (owned by root) during canonicalization of /run/systemd/netif.
Detected unsafe path transition / (owned by luk) → /var (owned by root) during canonicalization of /var.
Detected unsafe path transition / (owned by luk) → /var (owned by root) during canonicalization of /var/lib.
Detected unsafe path transition / (owned by luk) → /var (owned by root) during canonicalization of /var/lib/systemd.
Detected unsafe path transition / (owned by luk) → /run (owned by root) during canonicalization of /run.
Detected unsafe path transition / (owned by luk) → /run (owned by root) during canonicalization of /run/systemd.
Detected unsafe path transition / (owned by luk) → /run (owned by root) during canonicalization of /run/systemd.
Detected unsafe path transition / (owned by luk) → /run (owned by root) during canonicalization of /run/systemd.
Detected unsafe path transition / (owned by luk) → /run (owned by root) during canonicalization of /run/systemd.
Detected unsafe path transition / (owned by luk) → /run (owned by root) during canonicalization of /run/systemd.
Detected unsafe path transition / (owned by luk) → /run (owned by root) during canonicalization of /run/systemd.
Detected unsafe path transition / (owned by luk) → /run (owned by root) during canonicalization of /run.
Detected unsafe path transition / (owned by luk) → /var (owned by root) during canonicalization of /var/lib/systemd.
Detected unsafe path transition / (owned by luk) → /var (owned by root) during canonicalization of /var/lib.
Detected unsafe path transition / (owned by luk) → /var (owned by root) during canonicalization of /var/log.
Detected unsafe path transition / (owned by luk) → /var (owned by root) during canonicalization of /var.
Detected unsafe path transition / (owned by luk) → /var (owned by root) during canonicalization of /var/cache.
Detected unsafe path transition / (owned by luk) → /var (owned by root) during canonicalization of /var.
Detected unsafe path transition / (owned by luk) → /var (owned by root) during canonicalization of /var/log.
Detected unsafe path transition / (owned by luk) → /var (owned by root) during canonicalization of /var/log.
Detected unsafe path transition / (owned by luk) → /var (owned by root) during canonicalization of /var/log.
Detected unsafe path transition / (owned by luk) → /var (owned by root) during canonicalization of /var.
Detected unsafe path transition / (owned by luk) → /var (owned by root) during canonicalization of /var/log/journal.
Detected unsafe path transition / (owned by luk) → /var (owned by root) during canonicalization of /var/log/journal.
Detected unsafe path transition / (owned by luk) → /var (owned by root) during canonicalization of /var/log/journal.
Detected unsafe path transition / (owned by luk) → /var (owned by root) during canonicalization of /var/log/journal/2ef4ae7c728c43099e0f724fc5677ab7.
Detected unsafe path transition / (owned by luk) → /var (owned by root) during canonicalization of /var/log/journal/2ef4ae7c728c43099e0f724fc5677ab7.
Detected unsafe path transition / (owned by luk) → /var (owned by root) during canonicalization of /var/log/journal/2ef4ae7c728c43099e0f724fc5677ab7.
Detected unsafe path transition / (owned by luk) → /run (owned by root) during canonicalization of /run/log/journal.
Detected unsafe path transition / (owned by luk) → /run (owned by root) during canonicalization of /run/log/journal.
Detected unsafe path transition / (owned by luk) → /var (owned by root) during canonicalization of /var/log/journal/2ef4ae7c728c43099e0f724fc5677ab7/system.journal.
Detected unsafe path transition / (owned by luk) → /var (owned by root) during canonicalization of /var/log/journal/2ef4ae7c728c43099e0f724fc5677ab7/system.journal.
```
it outputed while upgrading system by ```Setting up systemd (252.26-1~deb12u2) ...```

hexdump0815, 2024-07-04:
thanks for the permission update - there was such a problem in the past already ( https://github.com/hexdump0815/imagebuilder/blob/main/doc/important-information.md#23-04-03-file-ownerships-can-be-wrong-on-aarch64-images ) which i hoped to got resolved, but looks in some form it seems to be back - i'll have to investigate in detail and if other images are affected aswell

sound testing: the problem is that there is a risk to burn the hardware or the ears - first rule: never with headphones directly on the ears as very loud or strange noises might come out ... in general: the audio mixers in such devices have often hundreds of settings and one needs to find the proper values for the proper settings to make audio work and if one sets the wrong ones to wrong values it can really burn the speakers due to too much power at the wrong frequencies (it had happend already a few times to some people) ... due to this its better to leave this area to people who really know what they are doing or have hw documentation

LukIsHere, 2024-07-04:
i don't think that's sth i'd be able to figure out, last thing i want to do is break this device
can't even open it to replace the speaker cause of some weird screws
sounds :3 like there isn't going to be any audio from the speakers any time soon

LukIsHere, 2024-07-05:
another issue i encountered is gdm3 always defaults to x11 no matter if i explicitly enable wayland in config or not
it's weird, it didn't happen on my other devices

LukIsHere, 2024-07-06:
another issue bluetooth doesn't work for input devices
bluetooth headphones are fine (as long as u don't need to pause or resume music with them)
bluetooth mouse doesn't work at all (it connects but doesn't do anything)
it is the case on kelner provided with image and [the newer kelner i've put together](https://github.com/hexdump0815/linux-mainline-and-mali-generic-stable-kernel/issues/15#issuecomment-2211492740)

almeida66, 2025-05-09:
Hi @thenameisluk,  I installed the last image, too. It starts/works from a sd-card; but the sd-port is quite slow.

I'm a noob with ARM: what about using the internal emmc as /root? Should it work and how to configure that on this image from hexdump0815?
Thx for a comment

thenameisluk, 2025-05-09:
@almeida66 

here is an **regualar** installation guide (assuming you aren't using testing image)
https://velvet-os.github.io/chromebooks/installation/basic-installation.html

if you use [**testing** image](https://github.com/hexdump0815/imagebuilder-testimages/releases/tag/spring-release-2025-testimages-round-01) (make sure it's variant corsola) there is a script for that
https://velvet-os.github.io/chromebooks/installation/script-installation.html
the script usually works on **regular** one but not in the case of your device

**testing** one has graphical glitches which need workaround before upgrading
putting 
```
PAN_MESA_DEBUG=noafbc
```
in ```/etc/environment``` should prevent them

almeida66, 2025-05-09:
Hi, 
found the older docs (install-to-emmc) behind the imagebuilder.tar. It worked for me.

btw. are there newer images, then the one from hexdump (2024-01)?

thenameisluk, 2025-05-09:
yes the testing images

https://github.com/hexdump0815/imagebuilder-testimages/releases/tag/spring-release-2025-testimages-round-01

almeida66, 2025-05-09:
hm, I was to impatient to read all stuff.... I looked for _corsola_ and was happy to find your thread.

well, maybe i try/install the newer image. Thx

archisman-panigrahi, 2025-09-04:
Is it possible to collaborate with https://github.com/WeirdTreeThing/chromebook-linux-audio for sound?

They fixed the sound issue in nearly all amd64 chromebooks back in the days when nothing worked (now some amd64 chromebooks have out of the box sound support in linux)

hexdump0815, 2025-09-05:
@archisman-panigrahi - no news yet in this area - someone needs to look deeper into this and maybe some kernel fixes might still be required - not sure ... the chromebook-linux-audio stuff might be good for learning about sof firmware and ucm files for sure where know how will be required to move forward

koen01, 2025-09-24:
I"m not sure this will help but I'm running postmarketos edge on this device (magneto) and with gnome the sound works for me. 
USB-c video didn't, but haven't tried USB-c to HDMI yet.

thenameisluk, 2025-09-26:
i've looked into it and it looks like the audio actually works on pmos corsole running 6.12.36-mt81 (so not colabora?)

i'll **try** looking into it but no promises

thenameisluk, 2025-10-01:
@Befaro2063 which image are you using?

thenameisluk, 2025-10-01:
weird i'll look into it

thenameisluk, 2025-10-01:
for whatever reason installing package via dpkg works
but with apt it doesn't

i've never seen that before

hexdump0815, 2025-10-01:
i created a pr for the mt81xx kernel repo at https://github.com/hexdump0815/linux-mainline-mediatek-mt81xx-kernel/pull/21 ... i'll merge it as soon as i have test-built and tested such a kernel and we maybe have more positive feedback about those patches working on corsola devices here

hexdump0815, 2025-10-04:
@thenameisluk @koen01 - i have created a separate gh issue to track the gnome/kde sessions not working here: https://github.com/velvet-os/imagebuilder/issues/381 - lets discuss that problem there from now on ... i'll also try to link it from a few other gh issues and discussions where it also came up

hexdump0815, 2025-10-04:
corsola magneton sound update 2: i have the sound now working on my corsola magneton with my v6.12.48 kernel based on https://github.com/hexdump0815/linux-mainline-mediatek-mt81xx-kernel/pull/21 (plus meanwhile https://github.com/hexdump0815/linux-mainline-mediatek-mt81xx-kernel/pull/22 as well, but that should not at all be sound related)

what i did to get there was:
- "mv /usr/share/alsa/ucm2 /usr/share/alsa/ucm2.off" and "rm /var/lib/alsa/asound.state" so that audio can no longer initialize in userspace properly resulting in no bad sound at bootup anymore, but also at no sound working at all anymore (as expected)
- booted up like this i did "mv /usr/share/alsa/ucm2.off /usr/share/alsa/ucm2" and restarted pipewire via "systemctl --user restart pipewire wireplumber pipewire-pulse"
- afterwards sound is working fine now, even after a reboot and also no more bad sound at bootup

i do not fully understand yet while it works this way and what was the problem - i guess somehow there must be somehow a wrong audio setup in the fresh images (but i have no idea where that should come from) ... i think more testing is required around this and in case someone else is doing so, please report here if you can reproduce this (i.e. bad sound when booting a new autum corsola image with a kernel with corsola audio fixes added and working sound after the procedure described above).

hexdump0815, 2025-10-15:
i think i found some usable workaround for the suspend/resume problem of my magneton and the nice thing is i found it by searching through the velvet-os github-issues, which are meanwhile quite a good ressource of information around linux on chromebooks :)

the idea for the workaround comes from https://github.com/velvet-os/imagebuilder/issues/60#issuecomment-2646256704 (suspend section) and after reading the linked page ( https://www.kernel.org/doc/Documentation/ABI/testing/debugfs-cros-ec ) plus some experimenting i realized that the solution used was most probably "echo 65535 > /sys/kernel/debug/cros_ec/suspend_timeout_ms" and with that applied on my magneton suspend/resumes works properly without any unexpected wakeup after a few seconds.

i would still be interested to hear from other magneton users, if they also see those suspend/resume problems - i think this could be due to some bug in the ec firmware which is there on some versions and not on others, resulting in some people having the problems and others not.

with this workaround now i think we have corsola at a similar supported and recommendable state as kukui and trogdor, which is very nice.

gwyddno, 2026-01-17:
That echo workaround also works for me so far, but doesn't seem to stick between restarts. If I don't run it after each login, my magneton will wake up after ~5-10 seconds, then reboot after another few seconds.

Not a complaint. I was wondering if that's just how it is at the moment, or if I'm doing something wrong on my end. I'm pretty new to chromebooks.

hexdump0815, 2026-01-18:
@gwyddno - happy that it works well for you now ... regarding testing: just using the device and pointing out any strange or unexpected issues here is enough already, so that we and others are aware of potential known problems

gwyddno, 2026-01-18:
@hexdump0815 

That was my thought as well about the Wayland issues. Seems like kernel 6.18 and mesa 25.1(? could be a slightly later version) include a lot of fixes and support for Bifrost and newer generation Mali graphics. 

I messed a little bit with a newer mesa version installed from trixie-backports with mixed results. I'm nowhere near brave enough, nor have the experience/knowledge to attempt getting kernel 6.18 going. 

I don't use X11 much these days unless to test against oddities on Wayland, but I'll update if anything changes or I find a workaround.

Glad to help/report my findings if they can be of any use.