velvet-os/imagebuilder #229: How to make Plymouth work.

GerardoHD, 2024-07-03:
So, does someone knows how to get Plymouth to work correctly? The other day by mistake I made it change from dmesg to Plymouth during boot and it only showed some white squares over a gray background. I searched on Google but everything seems to be oriented for GRUB+Plymouth.
Thank you in advance to whomever might answer.

GerardoHD, 2024-07-03:
Oh sorry, ill investigate that rebuilding process, hope i can get it to work!
Thank you, have a nice day

LukIsHere, 2024-07-21:
btw if u haven't figured that one out yet
if you do the luks installation (u can skip the luks part of it) it basically works without any issues
i think for normal install it just doesn't have the initramfs it needs but idk
https://github.com/user-attachments/assets/2b29093b-5cb0-4b30-ad66-40b9b449f977]
i made a custom one cause why not, it still needs a few tweaks

when come to it on my to do list will for sure make a tutorial or sth on setting it up

archisman-panigrahi, 2025-09-06:
Now that velvet-os shows a boot screen animation by default, how do we customize it? Is it as simple as installing a new plymouth theme?

thenameisluk, 2025-09-06:
@archisman-panigrahi 
1. install plymouth them
2. set it with
```
plymouth-set-default-theme [name]
```
3. rebuild initrams
```
sudo update-initramfs -u
```
4. flash kernel **if auto flash is disabled** (it's not by default)
```
sudo vtflash
```
or
```
sudo vttest
```
(for one time boot)

archisman-panigrahi, 2025-09-06:
Yay!

https://github.com/user-attachments/assets/eaba59a2-98cc-4254-a738-b49a9e5c3f31