velvet-os/imagebuilder #230: Which image will work for this Android TV box?

DealsBeam, 2024-07-04:
I believe this is the Amlogic S802, the PCB has "M9&M8_V1.0" on it, AIDA64 says SoC model: AMLogic Meson8 AML8726-m8, I believe it uses Broadcom network chip

https://imgur.com/SS2mkjU

https://imgur.com/xO06T1f

https://imgur.com/gth8WvC

I think this is the generic version: https://www.geekbuying.com/item/M8-Android-TV-BOX-Amlogic-S802-2G-8G-XBMC-USB-2-0-Android-4-3-Android-4-4-KitKat-OS-Black-324690.html

hexdump0815, 2024-07-04:
as stated in the readme: with tv boxes you are on your own - if any image works for that one then this one: https://github.com/hexdump0815/imagebuilder/tree/main/systems/amlogic_m8 ... just try all the meson8m2 and meson8 (not 8b) dtb files which can be used in /boot/uEnv.ini ( https://github.com/hexdump0815/imagebuilder/blob/main/systems/amlogic_m8/extra-files/boot/uEnv.ini ) and if you are lucky one of them will boot - if not then not

DealsBeam, 2024-07-04:
Thanks for the reply and your hard work.

I'm not that knowledgeable when it comes to Linux, commands, editing code files, etc. as I'd like to be. So all I have to do is keep the 'kver=KERNAL VERSION' the same and add/remove the # next to the dtb to try each one to see which one will boot? is this done when I put the SD card in the box and boot up? or is this done after the SD card is flashed?

Also, I've came across these https://linux-meson.com/ / https://github.com/devmfc/debian-on-amlogic / https://github.com/ophub/amlogic-s9xxx-armbian not sure if it can be helpful for developments.

DealsBeam, 2024-07-04:
So I downloaded the first image, unzipped the .gz file and flashed the extracted .img file using etcher and edited the uEnv.ini file and tried multiple dtb's - in the Android TV box, go to local update, select aml_autoscript.zip and hit update, the system restarts back into the Android TV box, not sure what I'm doing wrong.

Did I incorrectly flash the .img?

hexdump0815, 2024-07-05:
its some time ago i dealt with this - to me the steps you did sound correct - did you try to boot the image afterwards? it could be that that error might be ok, not sure.

if it does not work, you might have to search through the old forum threads and other places on the net how to do that initial step which is required to recognize linux systems to be booted from sd card and/or usb (not sure right now)

i'm aware of those other repos, but i have to admit, that i'm not any longer working really actively on amlogic tv boxes at the moment