velvet-os/imagebuilder #231: Cannot build bookworm and no wifi support

linsyking, 2024-07-06:
I used the script to build a chromebook_trogdor bookworm arm64 image.

However, I got stuck in the first step:

https://github.com/hexdump0815/imagebuilder/blob/main/scripts/create-fs-cache.sh

```bash
    wget https://ftp-master.debian.org/keys/release-12.asc -qO- | gpg --import --no-default-keyring --keyring ${DOWNLOAD_DIR}/debian-release-12.gpg
    LANG=C debootstrap --keyring=${DOWNLOAD_DIR}/debian-release-12.gpg --no-check-certificate --variant=minbase --arch=${BOOTSTRAP_ARCH} ${2} ${BUILD_ROOT_CACHE} http://deb.debian.org/debian/
    LANG=C debootstrap --variant=minbase --arch=${BOOTSTRAP_ARCH} ${2} ${BUILD_ROOT_CACHE} http://deb.debian.org/debian/
```

It seems that you `debootstrap` twice. The error is "cannot extract, file already exists" in the second `debootstrap`,

I removed the second `debootstrap` and build. When I boot the image built, there is no Wifi support (no wlan0 under /sys/class/net). Is this a firmware issue or some setting issue (udev?)

hexdump0815, 2024-07-07:
@linsyking - thanks a lot for spotting this debootstrap issue - it somehow looks strange and i'll have to look into this in detail ... it seems to be there for a while already, so no idea if something changed there or if i overlooked it all the time ... will look into it in the next days

no wifi on trogdor could be no firmware (there should be messages about it in dmesg most probably in that case) or missing qcom damons (see: https://github.com/hexdump0815/imagebuilder/blob/main/systems/chromebook_trogdor/postinstall-chroot.sh#L11-L12)

linsyking, 2024-07-07:
> @linsyking - thanks a lot for spotting this debootstrap issue - it somehow looks strange and i'll have to look into this in detail ... it seems to be there for a while already, so no idea if [...]

The firmware is OK. I used the same kernel and firmware to test. My old system could work properly but the new system cannot work. `dmesg` didn't show `ath10k_snoc` probing.

The device tree is loaded but the kernel isn't running the probe function.

linsyking, 2024-07-07:
I figured out that the problem is qcom damons.

It seems that `protection-domain-mapper` and `tqftpserv` are not necessary.

linsyking, 2024-07-07:
I found the problem was qcom damon.

From the documentation:

```
# install the qcom tools qrtr-ns and rmtfs
apt-get -y install protection-domain-mapper qrtr-tools rmtfs tqftpserv

# adjust the cmdline of rmtfs to read its files from /lib/firmware/rmtfs
# maybe this can be avoided by putting the dummy files from above to the
# default location /boot but some early tests did not work - might be worth
# to retest this
#sed -i 's,ExecStart=/usr/bin/rmtfs -r -P -s,ExecStart=/usr/bin/rmtfs -r -s -v -o /lib/firmware/rmtfs,g' /lib/systemd/system/rmtfs.service
```

I need to install all those four packages, but I only need to install `qrtr-tools rmtfs` to make it working. And I also need to change `/lib/systemd/system/rmtfs.service`.

What are `protection-domain-mapper tqftpserv`?

hexdump0815, 2024-07-07:
@linsyking - i think pd-mapper is meanwhile in the kernel )so the daemon is no longer required) and tqftpserv might not be required for trogdor if it works without it - i think it is used to load the firmware into the soc with something tftp like - it is needed for other qcom systems like msm8916 but newer ones do not need it anymore i think

hexdump0815, 2024-07-14:
@linsyking - it is hopefully fixed now with https://github.com/hexdump0815/imagebuilder/commit/116f742d027e98617f25a89084d94b646f31a8fd ... thanks a lot once more for spotting and reporting this issue