velvet-os/imagebuilder #233: Peach Pi/XE503C32: Speakers not working

Ashii-sh, 2024-07-12:
The latest image I used works so far, but I can't get the internal speakers to work on Peach Pi (XE503C32)
Audio works through the headphone output, but not through the internal speakers. I havent tested HDMI Audio, whether that works or not. 

It'll be appreciated if I can get some insight about this ! 

Image: Latest one for Pi Peach (https://github.com/hexdump0815/imagebuilder/releases/tag/230924-02) 

It *did* say "after first boot please run /scripts/fix-peach-audio.sh as root to fix the ucm audio config files for the corresponding system and reboot" on that page, but there wasn't such a script in that directory.

hexdump0815, 2024-07-12:
it looks like i removed that script at some point as it was not required for peach pit - maybe give it a try if it helps in your peach pi case: https://github.com/hexdump0815/imagebuilder/commit/5d314bc3f98244391d572f71c1c8e5e6c2c2ecb4#diff-925ce82269ebee50c5ac063a3a1f14a07c7a3b0062e27a3f6f6bd5ee571114b3 ... please let me know if it helps but i guess if you have headphones working, you kind of have working audio somehow already - did you already try to manually switch to speaker audio via pavucontrol?

Ashii-sh, 2024-07-13:
Okay so I tried the script, but it returns with "cp: cannot stat '/usr/share/alsa/ucm2/Peach-Pi-I2S-MAX98091': No such file or directory". 
And yeah, audio works except for the main speakers, and manually switching between both doesn't seem to make a difference in pavucontrol.

hexdump0815, 2024-07-14:
i have at least removed the note about the audio setup script now as it is just confusing if the script is not there anymore and not needed as it looks like: https://github.com/hexdump0815/imagebuilder/commit/f99b1a1b32e963501a7a05147bbc7742eb125f32

jenneron, 2024-07-15:
@hexdump0815 you can ship pmOS ucm configs as long as you ship the same kernel

hexdump0815, 2024-07-15:
@jenneron - thanks a lot, i'll note that down - actually iirc sound is working well on my peach pit - will have to check and retest all that whenever i find some time