velvet-os/imagebuilder #235: How to proceed once in terminal?

zetacripto, 2024-08-04:
Hi! I'm a total new to coding. Im trying to install the last image of Debian provided on this hub for Chromebook Nyan. The device I'm using is a Acer Chromebook 13 C810. 

So, I used rufus to install the image (https://github.com/hexdump0815/imagebuilder/releases/tag/230915-02) on a sd card. Then, after set the device on Dev Mode and do some stuff on Crosh to enable the usb boot, i managed to boot the usb.

Usual Linux terminal appeared, I log in on the root (root@changeme:~#), and then I tried to follow the steps on this guide (https://github.com/hexdump0815/imagebuilder/blob/main/doc/first-steps.md)

Then, I got totally lost. What should I do next?

Sorry for the inconveniences. I just want to make this laptop work for a person who only need to watch netflix and stuff. This is getting too advanced to me. Help!

thenameisluk, 2024-08-04:
@zetacripto 
the device should boot into a login screen, not terminal (afaik)

have you used the [nyan specific image](https://github.com/hexdump0815/imagebuilder/blob/main/systems/chromebook_nyan/readme.md)? or just downloaded a random on?

zetacripto, 2024-08-04:
> @zetacripto 
> the device should boot into a login screen, not terminal (afaik)
> 
> have you used the [nyan specific image](https://github.com/hexdump0815/imagebuilder/blob/main/systems/chromebook_nyan/readme.md)? or just downloaded a random on?

Hi! I used the image I specified on the issue. The page is: https://github.com/hexdump0815/linux-mainline-on-arm-chromebooks

[image]

Then I downloaded the first image on the list, since it was the most recent.

[image]

Maybe I'm doing something wrong in the process

thenameisluk, 2024-08-04:
@zetacripto sorry i missed that part.
at the bottom of the nyan page there are some notes
i found this
[image]

not sure if it's relevant to your model

either way i don't own any nyan chromebook so can't really help you much with that
if you don't find anything useful in notes there, you will have to wait for sb else to reply

zetacripto, 2024-08-04:
@thenameisluk Nah, my laptop has 4gb of RAM. Again, the model is Acer Chromebook 13 C810.

Maybe I installed wrong the image. I'm not sure also If I missed something in the process. 

However, in terminal I was able to check some things and was able to access to Debian (the root), so I really don't know.

Can you please give me info on how should be a normal installation process using one of this images?

thenameisluk, 2024-08-04:
@zetacripto 
you can try running ```startx``` command and seeing if it gives you any graphical environment

not sure if i get it, you want to install this image onto chromebook in a state it boots to rn?

if yes you want to install it **as is** check out [this tutorial](https://github.com/hexdump0815/imagebuilder/blob/main/doc/install-to-emmc-on-arm-chromebooks.md)

zetacripto, 2024-08-04:
@thenameisluk I was following these steps ok the first method (dd to emmc) but got lost on two things:

1. Why should I get a new image from terminal? I mean, download (wget), if I got already an image on my sd card (and I'm booting it)

2. I do not have wifi in terminal also to download that image xd.

Maybe I should follow other method? But wich one? 💀

I mean, there's a lot of stuff to do, and I do understand between 0 and nothing what I'm actually doing. 

Anyway, I'll try again to boot the usb.

zetacripto, 2024-08-04:
[image]

thenameisluk, 2024-08-04:
@zetacripto 
1. you can get wifi working (probably, if driver is there) with nmtui

2. there is rsync tutorial below that doesn't require internet

zetacripto, 2024-08-04:
@thenameisluk got stuck here[image]

zetacripto, 2024-08-04:
Ok, forgot to add a slash xd. Now I'm stuck here[image]

thenameisluk, 2024-08-04:
@hexdump0815 basic installation guide just asks for a rewrite or automated script 
@zetacripto could you provide an output of ```lsblk``` commmand?

zetacripto, 2024-08-04:
@thenameisluk [image]

Sorry for the inconveniences, and maybe I'm going too far. But can we use discord?

thenameisluk, 2024-08-04:
@zetacripto i'm fine with that
even [proposed discord server](https://github.com/hexdump0815/imagebuilder/issues/209) some time ago
but yeah currently there is no chatting platform
can do it in discord dm's in like 4 hours cause currently 'm busy

zetacripto, 2024-08-04:
> @zetacripto i'm fine with that
> even [proposed discord server](https://github.com/hexdump0815/imagebuilder/issues/209) some time ago
> but yeah currently there is no chatting platform
> can do it in discord dm's in like 4 hours cause currently 'm busy

@thenameisluk send me a friend request to: https://discord.gg/s78xgj4U

After that, maybe I can help to setup a discord server with all the documentation.

hexdump0815, 2024-08-04:
@thenameisluk - i just wanted to say a big "thank you" for all your support in various github issues here ... i cannot help myself right now due to real life (family) topics keeping me completely busy, but i at least try to follow the issues and try to add help if something comes tomy mind about the discussed topics - here the only thing would be the cgpt topic from https://github.com/hexdump0815/imagebuilder/blob/main/doc/important-information.md - not sure if it is relevant still and if in this case

zetacripto, 2024-08-05:
@hexdump0815 Hi! First of all, thanks for this project. Second, hope that you can resolve all the situations you're facing asap. 🙏

About the 32bit arm7l solution, it seems that could be the issue. However, it's my first time installing linux this way. There is something I should also know for my model? (Acer Chromebook 13 C810)

Meanwhile, I'll use rufus to installate the bullseye image and will keep reporting If something is not going as expected. 

Cheers!

hexdump0815, 2024-08-05:
@zetacripto - thanks a lot for thr good wishes ... i do not have a c810 myself, but i think someone got it working - see: https://github.com/hexdump0815/linux-mainline-tegra-k1-kernel/issues/5