velvet-os/imagebuilder #236: misc: Running windows applications in LXC containers using box64, box86, wine and lxd

linsyking, 2024-08-08:
Here is a guide running windows applications on archlinux arm64 PC installed by https://github.com/linsyking/imagebuilder.

Repo: https://github.com/linsyking/game-lxc-arm64

LXC works great. No problem with wayland and pulseaudio.

I tested running a few games and they seem to work great!

thenameisluk, 2024-08-09:
@linsyking 
it's nice you posted a guide <3

i don't see any reason why it wouldn't also work on debian (you know on of os'es this framework builds images for :3) or ubuntu

i'm not sure if u tested it there. if u haven't, i can try it myself later so you don't need to wipe your system just to do that

adding ```sudo apt install lxd``` for debian/ubuntu if it works wouldn't hurt

also i've put [some effort to extend the chromebook side of docs](https://github.com/hexdump0815/imagebuilder/pull/234) , if it goes through, the guide could be put in sth like doc/chromebooks/post/x.md

tech-with-mo, 2024-10-04:
Also try pi apps. Lots of ported apps of arm

thenameisluk, 2024-10-04:
pi apps don't really work well with trixie

just gonna end up with [FrankenDebian](https://wiki.debian.org/DontBreakDebian)

WesleyVanNeck, 2024-10-04:
> pi apps don't really work well with trixie
> 
> just gonna end up with [FrankenDebian](https://wiki.debian.org/DontBreakDebian)

Well pi apps steam doesn't work trogdor