velvet-os/imagebuilder #245: Restore simpler doc for emmc installation

archisman-panigrahi, 2024-09-24:
The [previous doc]((https://github.com/hexdump0815/imagebuilder/blob/3817d61b30069fe4541dac8612453ee34a597226/doc/install-to-emmc-on-arm-chromebooks.md)) for installing to emmc was much easier to follow than the [current doc](https://github.com/hexdump0815/imagebuilder/blob/main/doc/install-to-emmc-with-luks-full-disk-encryption.txt).

I am unsure if it was deleted by mistake or if it intentional, but it would be great if it is brought back.

I looked at the current version, and was totally unsure which path I should take (there seems to be one option for general chromebooks, one for EFI, one for u-boot, one for arm64 chromebooks, and so one). While that may be useful for advanced users, as someone cloning USB image to emmc for the first time, the previous version (now deleted) came really handy.

---
I am running Debian turned into Raspberry Pi OS on a Chromebook HANA, and everything (except GPU acceleration) is working. Thank you very much for maintaining this project.

thenameisluk, 2024-09-24:
@archisman-panigrahi 
### nothing was deleted
the previous doc is still in [doc/chromebooks/basic-installation.md](https://github.com/hexdump0815/imagebuilder/blob/main/doc/chromebooks/basic-installation.md)
you can find it by going [main readme](https://github.com/hexdump0815/imagebuilder) -> [special chromebook guide](https://github.com/hexdump0815/imagebuilder/blob/main/doc/chromebooks/preparing-chromebook.md) -> [regular installatin](https://github.com/hexdump0815/imagebuilder/blob/main/doc/chromebooks/basic-installation.md) (look at the bottom)

it was moved to chromebook specyfic doc since it's chromebook exclusive and the .txt one (you meantioned, explains the encrypted installation) wasn't since it wasn't just for chromebooks
you can actully find more friendly version i made of it at [doc/chromebooks/luks-installation.md](https://github.com/hexdump0815/imagebuilder/blob/main/doc/chromebooks/luks-installation.md) (chromebook only)

archisman-panigrahi, 2024-09-24:
@thenameisluk Thanks for the clarification. The link in first-steps.md was invalid, and I thought that the file has been deleted. I sent a PR.

PR #246 (2024-09-25): Fix missing links for eMMC installation
The links were not updated in first-steps.md, and I mistakenly assumed the tutorial has been deleted (#245).

Fixes #245
commit 83e37ee6 (2024-09-25): Fix missing links for eMMC installation (#246)

* Fix missing links for emmc installation
* Fixes #245
* add the word eMMC to preparing-chromebook.md

Also made some grammatical improvements
closed 2024-09-24 by archisman-panigrahi