velvet-os/imagebuilder #251: More Friendly Bootscreen?

tech-with-mo, 2024-10-04:
Is there a way for us to have a normal bootscreen like the Ubuntu logo, instead of the startup log?

thenameisluk, 2024-10-04:
@tech-with-mo yeah
one debian you can use [velvet tools](https://github.com/thenameisluk/velvettools) for that (what i use rn)

or that [luks installation script](https://github.com/hexdump0815/imagebuilder/blob/main/doc/chromebooks/kernel-and-initrd.sh)
velvet tools are based on (it's debian only too afaik)

i've already pulled it of
was even able to make a custom one
https://github.com/user-attachments/assets/2b29093b-5cb0-4b30-ad66-40b9b449f977

if you need a guide i can [i'm working on one](https://github.com/thenameisluk/imagebuilder/blob/main/luk's%20special%20todo)

tech-with-mo, 2024-10-04:
> @tech-with-mo yeah on debian you can use [velvet tools](https://github.com/thenameisluk/velvettools) for that (what i use rn)
> 
> or that [luks installation script](https://github.com/hexdump0815/imagebuilder/blob/main/doc/chromebooks/kernel-and-initrd.sh) velvet tools are based on (it's debian only too afaik)
> 
> i've already pulled it of was even able to make a custom one https://github.com/user-attachments/assets/2b29093b-5cb0-4b30-ad66-40b9b449f977
> 
> if you need a guide i can [i'm working on one](https://github.com/thenameisluk/imagebuilder/blob/main/luk's%20special%20todo)

Thank you so much