velvet-os/imagebuilder #252: Pen and touchscreen not working properly on kukui

BeNach0C, 2024-10-08:
My lenovo 10.1 duet boots and works with ubuntu just fine, but when using the pen and touchscreen they seem to be mapped like if the screen as on vertical orientation, even tho by default the screen is on landscape mode and that's how im using it.

Any fix for this?

thenameisluk, 2024-10-08:
@BeNach0C yeah that's a known issue
luckily it's solvable
look [here](https://github.com/hexdump0815/imagebuilder/issues/53#issuecomment-2001953008)

thenameisluk, 2024-10-08:
@BeNach0C not sure
you can try without

if it doesn't work
here is not merged yet [doc on changing de](https://github.com/thenameisluk/imagebuilder/blob/main/doc/chromebooks/post/switch-de.md)

BeNach0C, 2024-10-08:
Thank you very much, ill try it as soon as i can, ill also let you know if
it works with xfce.

Also as a side thing debian did not work on the tablet but im fine with
ubuntu.

Do you know if snap works?
I disabled because it does not work ok termux.

thenameisluk, 2024-10-08:
@BeNach0C 
weird debian didn't work for you, i've had no issues with debian on mine

what exactly happened? doesn't boot or sth?

no idea i don't use haven't used ubuntu image on my devices even once

BeNach0C, 2024-10-09:
Debian never booted, I've alwas installed them on external devices and
booted from them with a usb A dongle tho, ubuntu has always worked.
(Sorry for the late reply I've been really busy with uni)

hexdump0815, 2024-10-10:
@BeNach0C - what happened in the non-booting debian case - nothing or it hanging at some point while booting? if its the second, then try to shortly unplug and replug the usb device you are booting from - this seems to be required for some usb devices to be properly recognized during boot.

BeNach0C, 2024-10-10:
It simply beeped and put me back on the boot select screen. I know the
drive im using works because it boots ubuntu and postmarket.
I still preffer ubuntu tho so its not a problem for me, but would be nice
to know the issue if someone else needs debian.

thenameisluk, 2024-10-12:
without details the issue can't really be reproduced
the image works with my devices

it's likely sth with usb-c to usb adapter or similar

BeNach0C, 2024-10-12:
Im using 3 different type c to type a adapters, and none worked, but dont
worry ubuntu is working fine :)
Thanks for the help 🫂