velvet-os/imagebuilder #256: What is the usage of boot partition?

linsyking, 2024-10-15:
It seems that all the image, initramfs are written to the kernel partition, and the boot partition isn't used during booting, so why do we need to create one?

linsyking, 2024-10-15:
> well you don't really have to afaik
> unless you are doing luks encryption
> 
> but only in arm64 chromebook case
> there are other devices using uboot which require it (tv boxes/32bit chromebooks)
> 
> also /boot partition uses fat32 file system likely for some legacy reason (i think uboot again but not sure)
> 
> there are likely also some other best practice reasons
> 
> if you want to set it up without one you can try and on 64bit chromebooks it should work, but it's not 64 bit chromebooks only framework so it's likely better to keep stuff this way

Got it, thank you. I was trying to update my initramfs to use plymouth module and I add the initramfs to the kernel partition when building, but it seems that the system is booted without a initramfs. D you know how to use a custom initramfs? Thanks :)

linsyking, 2024-10-15:
I have changed the kernel cmdline from https://github.com/hexdump0815/imagebuilder/issues/99

But it doesn't work so i think we probably need to change the initramfs

thenameisluk, 2024-10-15:
you can use velvet tools
[here is guide](https://github.com/hexdump0815/imagebuilder/discussions/243#discussioncomment-10659997)

you can skip installing 6.11 part and just do
```
sudo apt install velvet-tools
```
and specify your preferred kernel version for vtbuild/vttest/vtflash

make sure that your ```/boot/cmdline``` generates with ```quiet splash``` after running vtbuild

if you have any issues let me know ;3

thenameisluk, 2024-10-15:
@linsyking did it work?
also i'm working on improving velvet tools so feedback would be appreciated

linsyking, 2024-10-15:
> @linsyking did it work?
> also i'm working on improving velvet tools so feedback would be appreciated

Sorry I am not running debian. Could you provide the source of those tools?

linsyking, 2024-10-15:
I see the tool [here](https://github.com/thenameisluk/velvettools/tree/mainhttps://github.com/thenameisluk/velvettools/tree/main). I wonder where is the option to specify the initramfs? So the initramfs must be fit into the 32M kernel partition along with the kernel image, is that right?

linsyking, 2024-10-15:
I'm wondering why even if we don't specify the initramfs, the system still boots succesfully? What initramfs does it use?

linsyking, 2024-10-15:
In my case I successfully boot the system with the initramfs and see the freeing message from `sudo dmesg | grep initrd`. However, the initramfs doesn't seem to have any effect.

thenameisluk, 2024-10-16:
> Sorry I am not running debian. Could you provide the source of those tools?

yeah the the tools are kinda debian only

it might be easier for you to try modify this [simpler script](https://github.com/hexdump0815/imagebuilder/blob/main/doc/chromebooks/installation/kernel-and-initrd.sh)

if you want you can port it to your system ([source code](https://github.com/thenameisluk/velvettools)) it's all bash

also about it not working
i think on arch you might have to add slash screen to initramfs manually but not sure

thenameisluk, 2024-10-16:
i'm not sure if you've seen that https://wiki.archlinux.org/title/Plymouth

linsyking, 2024-10-16:
> i'm not sure if you've seen that https://wiki.archlinux.org/title/Plymouth

Yes, I think the root cause the TTY problem. I can manually run plymouth by setting the tty option. I notice that the cmdline has a "console=tty1 console=ttyMSM0". May I ask what they are for?

thenameisluk, 2024-10-16:
for console on the screen

i know these can be set so you get serial console via usb using suzyq cable

thenameisluk, 2024-10-16:
https://github.com/hexdump0815/imagebuilder/discussions/243#discussioncomment-10674903

linsyking, 2024-10-16:
> for console on the screen
> 
> i know these can be set so you get serial console via usb using suzyq cable

Thank you. If I don't use suzyq cable, can I safely remove those options?

thenameisluk, 2024-10-16:
if you want you can remevo these but then you won't see kernel logs at boot (i think)

you can try