velvet-os/imagebuilder #258: [GUIDE] Getting Chromium Sync because there is no ARM version of Chrome.

tech-with-mo, 2024-10-26:
**GUIDE FOR CHROME SYNC ON OUR DEVICES**

For our ARM systems the only way to use Chrome, since Google never released an ARM version, is to install Chromium. Either through Flathub or the official Download. However, a few years prior Google got rid of Chromium Sync, which means no Bookmarks, Password and Cross-Device Sync. But there is a Solution. Pi-Apps! Pi-Apps is an awesome App Store built for the Pi, but Ubuntu as well as Debian is officially supported.
[Pi Apps](https://pi-apps.io/install/)

Install Pi-Apps: `wget -qO- https://raw.githubusercontent.com/Botspot/pi-apps/master/install | bash`
For an installation with a to see what's happening: `git clone https://github.com/Botspot/pi-apps && ~/pi-apps/install`

This should install the Pi-Apps GUI. **Note: before Proceeding you must know that a version of Chromium has to be installed.**
After the install is done open the Application and look for **Better Chromium** press install and you should get a Bash Window waiting for Authentification. Next you'll get a small Window asking for fixes to Apply including a few for better RAM and CPU usage as well as Application wide Dark Mode. After the installation is done you can open Chromium and Sign In. That's it.
If you wish to uninstall Pi-App open a Terminal and do: `~/pi-apps/uninstall`. This should uninstall the GUI and it's components.

This Guide can also be found on the [Pi-Apps GitHub Page](https://github.com/Pi-Apps-Coders/website/blob/master/src/install.md). If this Guide is not working on you're Device please feel free to edit the [Page](https://pi-apps.io/install/).

**Confirmed working on Chromebook Kukui, Snow and Peach**

thenameisluk, 2024-10-26:
be aware that this best works on stable debian

if you install these on unstable you might run into dep issues (and will create [FrankenDebian](https://wiki.debian.org/DontBreakDebian)

tech-with-mo, 2024-10-26:
For me I tried it on both Debian and Jammy (Tested it Stock after clean installation) and both worked perfectly, however this can change depending on you're config.

thenameisluk, 2024-10-26:
it's not about configuration

mismatching dependencies might cause issue
the store supports debian stable/old-stable and ubuntu
but not trixie

[image]
and it might cause issues if someone uses it there

tech-with-mo, 2024-10-26:
You're right! I didn't look at it that way. I'll update the guide! Thanks
for pointing this out!