velvet-os/imagebuilder #259: Help Needed with Debian 12 on Chromebook Snow rev4

Salahd94, 2024-10-26:
Hi I just got the bookworm Debian 12 destro kernal 6.1.11 on sd for my Chromebook snow rev4 and I have some issues.
1st. my mali gpu doesn't work and my sound also.
2nd. if I want just to remove the chrome os and setup the bookworm instead how can I do that?
ps. i'm not an expert so can your answer be in details thank u for all the work u've done

thenameisluk, 2024-10-27:
@Salahd94 
running ```/scripts/fix-snow-audio.sh``` should fix audio

also
"starting with the v6.1 kernels the legacy mali support has been dropped, so if it is really required it is recommended to use the v5.15 lts kernel which still has it and should be supported for quite some time as well"

you should look at some [system notes](https://github.com/hexdump0815/imagebuilder/blob/main/systems/chromebook_snow/readme.md)
you might find sth usefull

thenameisluk, 2024-10-27:
also when it comes to installing
there is [a guid](https://github.com/hexdump0815/imagebuilder/blob/main/doc/chromebooks/readme.md) on installing onto device memory

Salahd94, 2024-11-01:
Okay can you plz give me the download link to the Kernal 5.15 release coz it's not among the the releases
Thx🙏

thenameisluk, 2024-11-01:
installing new kernel might be tricky and i can't assist you with that yet (due to next to no experiance with uboot)

you can try using slightly older image with kernel 5.10.xx
https://github.com/hexdump0815/imagebuilder/releases/tag/210725-02

Salahd94, 2024-11-03:
In v5.10 release it runs on llvm and I can't change it whatever I did I can't make it run on the mali GPU so do u know how to like disable the llvm and make him run on the Mali GPU

Gredifa, 2026-10-26:
Hi team, first of all, thank you for the amazing work keeping these old Chromebooks alive. I'm using Velvet OS on my Samsung Chromebook Snow (xe303c12) Rev4.

**System Info:**
- Model: Samsung Chromebook Snow xe303c12 (Rev4)
- Architecture: armv7l
- OS: Velvet OS (Debian GNU/Linux 13 Trixie)
- Kernel: 6.12.92-stb-cbe+ #1 SMP PREEMPT
- Installation: Running from internal eMMC.

**The Issue:**
The internal audio is not working. The system does not detect any sound card.

**Diagnostics and Logs:**

1. `cat /proc/asound/cards` returns:
   `--- no soundcards ---`

2. `lsmod` shows the modules are loaded but maybe failing to bind:
   `snd_soc_max98095`, `snd_soc_snow`, `snd_soc_i2s`, etc.

3. `dmesg | grep max98095` shows the following errors right at boot (17:40:57):
   `max98095 7-0011: ASoC: error at soc_component_write_no_lock on max98095.7-0011 for register: [0x0000000f] -6`
   `max98095 7-0011: Failed to reset DSP: -6`
   `max98095 7-0011: Failure reading hardware revision: -6`
   `max98095 7-0011: ASoC: error at snd_soc_component_probe on max98095.7-0011: -6`
   `snow-audio sound: ASoC: failed to instantiate card -6`
   `snow-audio sound: error -ENXIO: snd_soc_register_card failed`

4. `sudo i2cdetect -y 7` (I2C bus for audio) shows address `0x11` (the MAX98095) is completely EMPTY (`--`), meaning the chip is not responding to I2C probes. Addresses `0x10`, `0x3b`, and `0x44` are detected.

5. Script issue: Running `/scripts/fix-snow-audio.sh` fails because the UCM directory (`/usr/share/alsa/ucm2/soundcards/HiFi.conf`) does not exist, likely because the card was never instantiated.

6. We also tried the community script from `WeirdTreeThing/chromebook-linux-audio`, but it explicitly says ARM Chromebooks are not supported.

**Hardware debugging:**
- `clk_summary` shows `mclk` IS enabled and running at 24MHz, bound to `7-0011` (the audio codec). So the clock is not the issue.
- Forcing GPIO 693 (gpiochip27, line 7) high using `libgpiod` (`gpioset -c gpiochip27 7=1`) does NOT wake the chip. The I2C address `0x11` remains dead.

**Conclusion:**
It seems the Device Tree (exynos5250-snow.dts) for the Rev4 hardware is missing the proper power-enable/reset GPIO configuration for the max98095. The mainline patch (e9eefc3f8ce09eb2262d7d817980816132158196) added the clocks, but perhaps for Rev4, the power sequence or the specific GPIO (EN/RESET) is different or missing in this kernel build.

Could you please check if the DTS for Snow Rev4 needs a specific patch to power on the max98095 codec? Is there a DTBO (Overlay) we can safely apply?

Thank you for your time and hard work!

hexdump0815, 2026-10-27:
@Gredifa - did you run the intial audio setup script in /scripts once + reboot? that is required to get audio working on snow and it should work on rev4 and rev5 and that scripts sets it up for the current device. i think this is also mentioned somewhere in the readme.md for snow ... if not, please let me know.