velvet-os/imagebuilder #260: Screen goes black with a single underscore

thunderclaw09, 2024-10-27:
I'm running this on an M8S with a S812 processor, and once everything loads in with the green OK, the screen goes blank with a single underscore. Typing on the keyboard does nothing, and leaving it does nothing. The last entries before this happens are "starting lightdm.service" and "started systemd-hostnamed.service"

thenameisluk, 2024-10-27:
@thunderclaw09 sorry if it actually doesn't work we can't help you

you might try swapping dtb's and maybe specyfying your [tv resolution in cmdline](https://github.com/hexdump0815/imagebuilder/issues/209#issuecomment-2408693839)

but like you can see [here](https://github.com/hexdump0815/imagebuilder/blob/main/systems/amlogic_m8/readme.md)
[image]
we can't support all of them since some are just faker

thenameisluk, 2024-10-27:
if screens goes black it's and your device freezes
it might be under volted

you might want to try dtb's with some additional power if any are available