velvet-os/imagebuilder #264: Creating images with custom system

Eeeevox, 2024-11-08:
Is there a way to just dump ChromeOs kernel from a prebuild iso for your system and just replace all the other partitions and point the kernel to them?

I'm wondering couz i'm not a big fun apt and outdated packages and just think that a guide or even a script that would to this automaticlly would be supper useful. I can help with the script the problem is that i only know java and python so yeah

thenameisluk, 2024-11-08:
@Eeeevox using chromeos kernel will likely cause gpu drivers issues

have you ever heard of debian trixie/testing it's basically rolling release but if you need something even more bleeding edge you can use debian sid

if you still want to really badly to use other linux there are projects bringing arch to kukui and trogdor chromebook

if you want some more exotic distro either read the srcipts the were used to create this images that are literally in this repo (it's not hard to modify them, it's just bash) or get one of the images and bootstrap your distro of choice onto 4th partition and call it a day

if you have any additional questions let mw know

Eeeevox, 2024-11-08:
@thenameisluk i've tried to bootstrap my own distro but i cant quit figure out how

I copied the first and second (labeled Chrome Os kernel) partitions from the image with gnome disks and i cant get the partitions to write or even read, can you recommend me any software to do that?

thenameisluk, 2024-11-08:
@Eeeevox 
bruh
just flash the image we provide
```
dd if=image of=/dev/sd? status=progress
```
_Tip. replace ? with letter of your flashdrive_

mount root partition
```
mount /dev/sd?4 /mnt
```
then clear it
```
rm -rf /mnt/*
```

bootstrap onto /mnt

chroot there and set up fstab to mount partition LABEL=rootpart in / and LABEL=bootpart in /boot

remember to run
```
sudo sync
```
before unplugging any storage device

and under not circumstances should you play with partitions (remove and create them) if you don't know what you are doing

make sure your distro has an arm version and that you bootstrap it

if you are unable to do it please reconsider using whatever you want to use, just because "cool kids" use it, it doesn't mean you have to use it too
especially if you have no idea what you are doing