velvet-os/imagebuilder #270: No hardware acceleration (i think)

awesomefroggy29, 2024-12-06:
I am currently using an IdeaPad Flex 3 CB 11M735 (C330) on a mediatek mt8173 and for some reason, any sort of game i run is extremely slow and laggy. But if i use crostini on chromeos, the same game runs much much better. Is there any way to fix this?

thenameisluk, 2024-12-06:
@awesomefroggy29 
sorry but ur device doesn't currently have gpu acceleration

well the issue is chrome os uses proprietary gpu driver which enables gpu acceleration, you might be in theory able to get that driver working on ur device but i've never tried

there is an [open sorce version](https://docs.mesa3d.org/drivers/powervr.html) in development but noone here really tried to get it working even thoe it should be in theory compatible with ur device https://wiki.postmarketos.org/wiki/Lenovo_IdeaPad_Flex_3_Chromebook_(CB-11M735)_(google-hana)

thenameisluk, 2024-12-06:
i got my hands on the device through means

will see what can be figured out with it currently

awesomefroggy29, 2024-12-06:
ok thanks! ill try to figure it out...

awesomefroggy29, 2024-12-06:
yeah... its not working. I can't even get the first step in installing to work.

hexdump0815, 2024-12-06:
it will still take some time until the powervr open source gpu driver will be useable - it is slowly getting there, but i would not expect anything useable at all before mid or end of next year

awesomefroggy29, 2024-12-06:
oh, i can wait lol. Ive been trying to get linux on my chromebook for a year and a half now, and i finally found this(i have no idea how it took so long) so yeah, i can wait.

awesomefroggy29, 2024-12-12:
i think im getting somewhere... ive watched a video and it seems to work so far, ill update on it later

hexdump0815, 2024-12-12:
oh - you mean video decoding accel - that might even work with mt8173 (it is not using the gpu, but the vpu = video processing unit i think) - i was assuming gpu 3d accel :)

awesomefroggy29, 2025-01-28:
yes, the GPU 3d accel is what I'm looking for

awesomefroggy29, 2025-04-04:
haha, @hexdump0815, I am still waiting lol. I have looked into the ChromeOS shell a lot since January and used the emerge command to install packages on my device. they only problem is uhh... the portage(gentoo package manager) version and other things that have to do with portage are either too old, or just missing overall. so when I add a repo, It just uhh, doesn't work. so I am back to this. any progress on the hardware accel?

hexdump0815, 2025-04-04:
sadly nothing i'm aware of ... one might try what is there in the powervr devel tree of mesa with zink but i doubt that it will be useable in any way right now ... the powervr kernel driver is in mainline maenwhile and the firmware file for the mt8173 gpu is also in some powervr firmware tree iirc

awesomefroggy29, 2025-04-05:
@hexdump0815 is there like any sort of link to download the driver? or the source code that i can build?

hexdump0815, 2025-04-05:
https://gitlab.freedesktop.org/frankbinns should give you some pointers - but do not expect to to be in a state to just compile it and it works - there will be quite a bit of research required, maybe creating an issue there asking for the state for the mt8173 and be ready that it still is not useable at all on this plaform ... i think they have it working for some texas instruments boards (who paid for getting it working i think) to some degree and everything else is still wip ... if you find out anything new, please let us know here ...

awesomefroggy29, 2025-04-05:
i got it to work again, but xfce is kind of broken

awesomefroggy29, 2025-04-05:
@hexdump0815 so basically what i did is compile the drivers from the guy at the top of this issue. then i installed them to the system and rebooted, but then the x server didnt start. so i removed all of the drivers and now xfce4 is sort of broken, but i can fix it just fine.

hexdump0815, 2025-04-06:
as said: just compiling some branch from some repo will most probably not get you very far - it is required to understand how everything works together and hten try to get it working ... just some things which come to mind:
- is the mainline powervr kernel driver coming up well and reading the proper firmware?
- is mesa compiled with the proper driver for it and is the id of the gpu enabled for it?
- the powervr driver is vulkan only, so there will be no opengl/gles support with it out of the box and znk will be required for that - so for testing things like vulkan testing tools would be the correct choice (vkcube etc.?)
- if things are not working at all or only partially, discussing this in a to be opened issue on the powervr repo might be required as support from the powervr developers might be required to get things forward or at least they should be able what is missing and when it might be available

awesomefroggy29, 2025-04-07:
@hexdump0815, do you think it would work if I copied the drivers from chromeos to Linux?

thenameisluk, 2025-04-07:
which part of

> it is required to understand how everything works together and then try to get it working

don't you understand

chromeos doesn't even use mesa driver (which debian uses) so no it **wouldn't just work** unless you also get a chromeos kernel and set it all up so your DE actually can use ur funny closed source driver

we don't currently work on getting it working because it's till in development if you have any questions about it you should ask around in the gitlab you were give (https://gitlab.freedesktop.org/frankbinns) we can't help you here since we don't currently work on it

hexdump0815, 2025-04-07:
getting the powervr driver working is really a quite heavy topic and the only thing one can try is to do some research, experiments and maybe **constructively** (i.e. be giving very detailed debugging and test feedback) helping the few people (frank binns and team) getting it working step by step on more systems ... it seems to be working on some texas instrumtens sbc i think (and even there it still has lots of bugs) and all other pvr gpus are more or less different from the one used there, i.e. require additional code which only the pvr team can bring in as only they have access to the required documentation ... a good sign of how complicated this topic is, is that nearly all risc-v sbcs of the last years which have a gpu are using pvr gpus and none of them has a working open source mesa pvr driver yet and there are paying companies and millions and millions of sold boards behind ...

awesomefroggy29, 2025-04-07:
@hexdump0815 and @thenameisluk, i have created an issue in gitlab.freedesktop.org/mesa/mesa for the mt8173. hopefully someone responds and with enough luck i can get it to work. tbh i dont really know why i care about this so much. but a challenge is a challenge.

hexdump0815, 2025-04-08:
just for reference - this is ther gitlab issue you mentioned: https://gitlab.freedesktop.org/mesa/mesa/-/issues/12959 ... i'm not expecting much to happen around it as without any detailed debugging and status info nobody will really care most probably - but lets see, maybe someone will add something to it or others who have experimented around it might speak up

awesomefroggy29, 2025-04-08:
@hexdump0815 yup, that's it.

awesomefroggy29, 2025-04-08:
@thenameisluk one more thing. Now I have no idea how crostini works with the host GPU, but could I copy the drivers from crostini? I am using the debian install that hexdump supports, so it _could_ work, but I don't know. I also just compiled a different version of mesa, 24.1.7 instead of 25.0.3, but now I think I completely broke my install, as when I boot up, there is just an underscore at the top left and it wont go away. even if I press CTRL + ALT + F2 or F3, etc.

hexdump0815, 2025-04-08:
crostini approach will not work neither as it uses some kind of gpu virtualization layer between the host and vm which uses the chromeos gpu driver as a base

i think there are only two ways to get to a working gpu on mt8173: patience until someone gets it working or learning how it works (a lot to learn) and try to work together with the pvr devs to get it working (not sure if they would be willing to help with it) ... i'm not sure which approach would give a result first :)

awesomefroggy29, 2025-04-08:
alright.

awesomefroggy29, 2025-05-04:
@hexdump0815 you wouldn’t believe it. Someone actually commented saying something about using mesa 24.3.4 instead of 24.1.7. They said it worked on their Lenovo N23 Yoga Chromebook with the same processor as mine, and they were running postmarketos. You can check it out again for yourself. I’m going to try it soon

hexdump0815, 2025-05-04:
@awesomefroggy29 - if you get anything working, please let us know - do you have a link for the pmos success info?

awesomefroggy29, 2025-05-04:
I'm not sure what pmos is.

Thunder-ai, 2025-05-04:
Pmos is magic, what do you want to know about it?

awesomefroggy29, 2025-05-05:
im not sure, i just want to know what hexdump means by pmos success info

Thunder-ai, 2025-05-05:
Oh, mb.

awesomefroggy29, 2025-05-05:
ah ok. you can find what they said here: https://gitlab.freedesktop.org/mesa/mesa/-/issues/12959 

if youre asking the success info from me, i am running the debian image from https://github.com/hexdump0815/imagebuilder/releases/tag/230920-01. so i cant give any.

Thunder-ai, 2025-05-05:
I'll have to try that, lemme know if yours works. (Imma also see if trixie will work, but Elan touchpad is wonky for trixie, i'll also try Pmos if neither work.)

awesomefroggy29, 2025-05-05:
@hexdump0815 so far, i am just getting all of the dependencies for building mesa, and everything seems to be working just fine, just the normal setup for building mesa.

awesomefroggy29, 2025-05-05:
also, when i am done and if this works, how do i send you the zip file of the folder with the built files? can i do it over google drive?

awesomefroggy29, 2025-05-06:
@Thunder-ai i just finished building and compiling mesa 24.3.4, heres the link if you would like to install it with `sudo meson install -C builddir/` just warning you, this may break your os that you're running it on as when i tried other versions of mesa, i had to reinstall debian. so i would backup your files first.

Thunder-ai, 2025-05-21:
@awesomefroggy29 Will try and update if working.

thenameisluk, 2025-05-21:
just clone this branch https://salsa.debian.org/xorg-team/lib/mesa/-/tree/mesa-24.3.4-3?ref_type=tags
and run
```
sudo apt build-dep .
dpkg-buildpackage --build=binary
```
to build .deb package

this way you don't risk breaking your system with meson's install script

awesomefroggy29, 2025-05-22:
@thenameisluk so compiling from https://docs.mesa3d.org/drivers/powervr.html did work as I showed in a previous comment but just installing it doesn’t really do anything. At least it doesn’t break the system like before, but not much progress for that part.

hexdump0815, 2025-05-23:
@awesomefroggy29 - reading through https://gitlab.postmarketos.org/postmarketOS/pmaports/-/merge_requests/6465 to me it looks like the support is still not at a point where it is useable: kernel driver seems to probe (with extra kernel patches), mesa (built with powervr support enabled, which seems to be disabled by default in debian still) can be compiled for it and vulkaninfo works and thats it so far anything further seems to not yet work ... so it might be better to wait a bit more until there is some more progress there or you really have to learn a lot on yourself from the ressources in the net how to compile a kernel with an added patch (which might require adjustments for a certain kernel version etc.) and how to compile mesa with non-standard options and make sure your system is really using the self compiled mesa version and all that without breaking your system ... i do not want to discourage you - i just want to say that for the gpu on mt8173 there is currently no easy way to get it working as there is still a lot more work required to get it working

awesomefroggy29, 2025-05-23:
Ok

hexdump0815, 2025-05-24:
@awesomefroggy29 - the kernel patch you can extract from the postmarketos merge request - powervr-MT8173-GPU-support.patch

awesomefroggy29, 2025-09-13:
@hexdump0815 what device tree blob is the device loading? i have no idea what it is, and i kinda need it right now to progress

awesomefroggy29, 2025-09-13:
if you have it, do you have the pvr.ko or similar? i am trying to figure out where i can get it

awesomefroggy29, 2025-09-13:
@hexdump0815 so good news, i built linux kernel 6.8 with the config in /boot, and added imagination technologies, and booted into it with kexec, and it booted! trying to make it permanent so i can use it from now on...

awesomefroggy29, 2025-09-14:
im applying the patch and building

thenameisluk, 2025-09-14:
do you include the firmware in initramfs?
it might be causing issues if you don't

awesomefroggy29, 2025-09-14:
i think i tried both, and it still did that. either way the service i created works just fine.

Thunder-ai, 2025-12-18:
ChromeOS is based* on linux, but with a whole lot of the typical stuff nuked from orbit, i.e. pretty much all the GNU userland stuff, so theoretically it can be deforked, there is a big, massive issue though, it's proprietary, meaning, i am clueless how this would work with the GPL, if it is even permitted by the GPL to have it proprietary, or if the ChromeOS kernel is even under the GPL. That said, i believe it is technically possible, i just wouldn't redistribute anything just yet. *Based can mean it has 50% original code, and 50% google code that makes it incompatible.

awesomefroggy29, 2025-12-19:
> But note that i didn't say you couldn't screw around with the code to see if it works on Debian (Suggested due to Crostini), i skimmed through the GPL, it "Should" be fine to try and get the drivers working, and maaaybe redistribute them, but i am no legal expert.

if you look here:
https://chromium.googlesource.com/chromiumos/third_party/kernel/+/chromeos-3.18/arch/arm64/boot/dts/mediatek/mt8173-oak.dtsi

it says:
<html>
<body>

/*
--
* This program is free software; you can redistribute it and/or modify
* it under the terms of the GNU General Public License version 2 as
* published by the Free Software Foundation.
*
* This program is distributed in the hope that it will be useful,
* but WITHOUT ANY WARRANTY; without even the implied warranty of
* MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
* GNU General Public License for more details.
*/

</body>
</html>

at the top, so its most likely legal to redistribute this.

awesomefroggy29, 2025-12-19:
we might finally be able to close this issue lol

hexdump0815, 2025-12-19:
@awesomefroggy29 - chromeos uses some very old hacked together mesa to support the legacy binary blob libs from img ... current state of the open source pvr driver is: kernel part is there, user-space/mesa part is being worked on slowly, but they no longer work actively on the gpu used in oak, but instead focus on newer ones, i.e. if support for the oak gpu will ever make it into mainline mesa it will have to come from the community ...

awesomefroggy29, 2025-12-23:
ok i also kind of got zink to work

awesomefroggy29, 2025-12-23:
i opened doom 3 demo from pi apps, faster than llvmpipe, but then freezes

hexdump0815, 2025-12-23:
@awesomefroggy29 - this is with the old hacked mesa or with the current mainline stuff? interesting progress anyway ...

awesomefroggy29, 2025-12-23:
@hexdump0815 with current mainline mesa. i managed to stuff a bunch of flags into the userspace through bashrc:
`export LD_LIBRARY_PATH=/usr/local/lib/aarch64-linux/gnu:$LD_LIBRARY_PATH`
`export VK_X11_FORCE_DRI3=1`
`export VK_ICD_FILENAMES=/usr/local/share/vulkan/icd.d/powervr_mesa_icd.aarch64.json`
`export MESA_LOADER_DRIVER_OVERRIDE=zink`
`export GALLIUM_DRIVER=zink`
`export MESA_VK_WSI_DEBUG=sw #due to dri3 support being absent, still trying to find a workaround for this to force it.`
`export LIBGL_KOPPER_DRI2=1`
`export MESA_VK_WSI_PRESENT_MODE=mailbox`
`export PVR_I_WANT_A_BROKEN_VULKAN_DRIVER=1`
`export vblank_mode=0`

and editing some mesa files to force allowing a window to open with it. ex:
~/mesavr/src/vulkan/wsi/wsi_common_x11.c:421:
from:
`static bool`
`wsi_x11_check_for_dri3(struct wsi_x11_connection *wsi_conn)`
`{`
`   if (wsi_conn->has_dri3)`
`      return true;`
`   if (!wsi_conn->is_proprietary_x11) {`
`      mesa_logi("vulkan: No DRI3 support detected - required for presentation\n"`
`                "Note: you can probably enable DRI3 in your Xorg config\n.");`
`   }`
`   return false;`
`}`

to:
`static bool`
`wsi_x11_check_for_dri3(struct wsi_x11_connection *wsi_conn)`
`{`
`   if (wsi_conn->has_dri3)`
`      return true;`
`   if (!wsi_conn->is_proprietary_x11) {`
`      mesa_logi("vulkan: No DRI3 support detected - required for presentation\n"`
`                "Note: you can probably enable DRI3 in your Xorg config\n. Attempting anyway because this is most likely a test.");`
`   }`
`   return true;`
`}`

and there are more things i had to change, and using mesa 26.0.0-devel(mainline), although im pretty sure you could also use 25.3.2.

awesomefroggy29, 2025-12-23:
ok that example that i listed above was actually the fix for dri3, hopefully i can figure out how to get gnome working with the gpu now. also, just so you know, vkcube works, vkmark works, glmark2 works but then segfaults after the fourth test

Thunder-ai, 2026-03-14:
I tried the kernel patch @awesomefroggy29 , i think it worked, it showed up in sys/modules,  but the chomeos kernel (5.15) either doesn't boot, or miss-init's the gpu, and black screen (usb heats up, so i suspect GPU driver problems), using the regular kernel, i force pointed glxgears at the driver, but no meaningful speed increase. I may need to see if i might be just a build (i.e 6.12.668 vs 6.12.676) too high.

awesomefroggy29, 2026-03-14:
you need quite a bit more than just pvr_dri.so. also before you waste your time, the vulkan driver from chromeos wont work(even though you see it) as i checked with vulkaninfo from portage, and it doesnt have any of the extensions that x11 or wayland needs, yet alone any program besides surfaceless ones. other than that, the opengl driver seems to be conformant in chromeos. if you copy over every opengl file, the linker(yes, the linker is needed from chromeos unfortunately), and the libc from /lib64, you MAY and i say MAY get it working.

hexdump0815, 2026-11-09:
just fyi: this nura (formerly known as postmarketos) workitem describes the current state of the powervr gpu driver on oak quite well: https://gitlab.nura.eco/nura/pmaports/-/work_items/4832

in summary: it is currently not really useable and its unclear if it will ever be ...

[1 more reply]