velvet-os/imagebuilder #274: Acer tab 10 black screen after no operation for a while

goddade, 2024-12-15:
After turning off all power save options, the screen will still be black after a while and cannot be restored.

hexdump0815, 2024-12-15:
do you see anything in dmesg when this happens (assuming you can login remote to the machine to see the kernel logs still)?

WesleyVanNeck, 2024-12-17:
I got this on kde but gnome is fine

hexdump0815, 2024-12-17:
@WesleyVanNeck @goddade - is this with x11 or wayland?

WesleyVanNeck, 2024-12-20:
> @WesleyVanNeck @goddade - is this with x11 or wayland?

 for me its on  wayland

goddade, 2024-12-22:
The image I use is from https://github.com/hexdump0815/imagebuilder/releases/tag/231001-01

archisman-panigrahi, 2025-02-11:
I am thinking about getting this device. Is the camera supported in Debian/Ubuntu? Does the microphone work?

thenameisluk, 2025-02-11:
@archisman-panigrahi camera doesn't work on any tablet chromebooks
only laptops