velvet-os/imagebuilder #281: ℹ️ [GUIDE]: Getting Rain World (and wine) on arm64 linux

thenameisluk, 2025-01-10:
[image]

_Note. if you only want wine, just use [hangover](github.com/AndreRH/hangover)_

but since hangover doesn't work with Rain World at them moment, here is how to get it working
(this might also work for other Windows games, no promises)
_Note. this seams to only work for devices with adreno gpu since you need directx11 but might work for mali (see balow)_

this setup is for [GOG](https://www.gog.com/en/game/rain_world) version of rain world since setting up steam is too hard

1. we add armhf architecture
```
sudo dpkg --add-architecture armhf
sudo apt-get update
```
2. install some deps
```
sudo apt-get install libc6:armhf mesa*:armhf zenity:armhf
```
3. install box86
```
sudo wget https://ryanfortner.github.io/box86-debs/box86.list -O /etc/apt/sources.list.d/box86.list
wget -qO- https://ryanfortner.github.io/box86-debs/KEY.gpg | sudo gpg --dearmor -o /etc/apt/trusted.gpg.d/box86-debs-archive-keyring.gpg 
sudo apt update && sudo apt install box86-rpi4arm64 -y
```
4. install box64 (likely optional for this)
```
sudo wget https://ryanfortner.github.io/box64-debs/box64.list -O /etc/apt/sources.list.d/box64.list
wget -qO- https://ryanfortner.github.io/box64-debs/KEY.gpg | sudo gpg --dearmor -o /etc/apt/trusted.gpg.d/box64-debs-archive-keyring.gpg 
sudo apt update && sudo apt install box64 -y
```
5. download wine binaries
```
wget https://dl.winehq.org/wine-builds/debian/dists/buster/main/binary-i386/wine-stable-i386_6.0.2~buster-1_i386.deb
wget https://dl.winehq.org/wine-builds/debian/dists/buster/main/binary-i386/wine-stable_6.0.2~buster-1_i386.deb
wget https://dl.winehq.org/wine-builds/debian/dists/buster/main/binary-amd64/wine-stable-amd64_6.0.2~buster-1_amd64.deb
wget https://dl.winehq.org/wine-builds/debian/dists/buster/main/binary-amd64/wine-stable_6.0.2~buster-1_amd64.deb
```
6. unpack them and move the files
```
for deb in ./*.deb ;do  dpkg-deb -xv $deb wine-installer; done
rm *.deb
mv wine-installer/opt/wine* ~/wine
rm -rf wine-installer
```
7. symlink binaries
```
sudo ln -s ~/wine/bin/wine /usr/local/bin/wine
sudo ln -s ~/wine/bin/wineboot /usr/local/bin/wineboot
sudo ln -s ~/wine/bin/winecfg /usr/local/bin/winecfg
sudo ln -s ~/wine/bin/wineserver /usr/local/bin/wineserver
sudo chmod +x /usr/local/bin/wine /usr/local/bin/wineboot /usr/local/bin/winecfg /usr/local/bin/wineserver
```

now you should be able to install rain world by running ```wine rain_world_gog_installer.exe```
_Note. the name is likely diffrent_

after installation, the rain world should be located at ```~/.wine/drive_c/GOG Games/Rain World```

(inside that directory) you can run it with
```
BOX86_DYNAREC_STRONGMEM=1 wine RainWorld
```
_Note. BOX86_DYNAREC_STRONGMEM prevents game from crashing for no reason_
_Note. on devices with mali gpu adding ```MESA_GL_VERSION_OVERRIDE=3.2 PAN_MESA_DEBUG=gl3``` **might** make it possible to run the game_

Tested on lenovo ideapad duet 5 (homestar)

Based on [this](https://forum.armbian.com/topic/19526-how-to-install-box86-box64-wine32-wine64-winetricks-on-arm64/)
working on op6t is a bit overwhelming hence the tutorial for me to chill out

WesleyVanNeck, 2025-01-10:
> ![rw](https://private-user-images.githubusercontent.com/92756992/401740565-dd82d4af-f2c7-4f9e-acd6-030ea25257a7.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY1MzA1MDUsIm5iZiI6MTczNjUzMDIwNSwicGF0aCI6Ii85Mjc1Njk5Mi80MDE3NDA1NjUtZGQ4MmQ0YWYtZjJjNy00ZjllLWFjZDYtMDMwZWEyNTI1N2E3LmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTEwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDExMFQxNzMwMDVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zZjllZWExZGJlZTBhZmQzNDk4MGRhN2I5NzdkM2RmMmM3NThiZWI1Mzg1YTQ0MjgxZDhlNmI0YzIzOTgxZTBkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.2hiSM0TC3EciwIvid7mBeJlygRluFE6f9qBS6sTIpes)
> 
> _Note. if you only want wine, just use [hangover](github.com/AndreRH/hangover)_
> 
> but since hangover doesn't work with Rain World at them moment, here is how to get it working (this might also work for other Windows games, no promises) _Note. this seams to only work for devices with adreno gpu since you need directx11 but might work for mali (see balow)_
> 
> this setup is for [GOG](https://www.gog.com/en/game/rain_world) version of rain world since setting up steam is too hard
> 
>     1. we add armhf architecture
> 
> 
> ```
> sudo dpkg --add-architecture armhf
> sudo apt-get update
> ```
> 
>     2. install some deps
> 
> 
> ```
> sudo apt-get install libc6:armhf mesa*:armhf zenity:armhf
> ```
> 
>     3. install box86
> 
> 
> ```
> sudo wget https://ryanfortner.github.io/box86-debs/box86.list -O /etc/apt/sources.list.d/box86.list
> wget -qO- https://ryanfortner.github.io/box86-debs/KEY.gpg | sudo gpg --dearmor -o /etc/apt/trusted.gpg.d/box86-debs-archive-keyring.gpg 
> sudo apt update && sudo apt install box86-rpi4arm64 -y
> ```
> 
>     4. install box64 (likely optional for this)
> 
> 
> ```
> sudo wget https://ryanfortner.github.io/box64-debs/box64.list -O /etc/apt/sources.list.d/box64.list
> wget -qO- https://ryanfortner.github.io/box64-debs/KEY.gpg | sudo gpg --dearmor -o /etc/apt/trusted.gpg.d/box64-debs-archive-keyring.gpg 
> sudo apt update && sudo apt install box64 -y
> ```
> 
>     5. download wine binaries
> 
> 
> ```
> wget https://dl.winehq.org/wine-builds/debian/dists/buster/main/binary-i386/wine-stable-i386_6.0.2~buster-1_i386.deb
> wget https://dl.winehq.org/wine-builds/debian/dists/buster/main/binary-i386/wine-stable_6.0.2~buster-1_i386.deb
> wget https://dl.winehq.org/wine-builds/debian/dists/buster/main/binary-amd64/wine-stable-amd64_6.0.2~buster-1_amd64.deb
> wget https://dl.winehq.org/wine-builds/debian/dists/buster/main/binary-amd64/wine-stable_6.0.2~buster-1_amd64.deb
> ```
> 
>     6. unpack them and move the files
> 
> 
> ```
> for deb in ./*.deb ;do  dpkg-deb -xv $deb wine-installer; done
> rm *.deb
> mv wine-installer/opt/wine* ~/wine
> rm -rf wine-installer
> ```
> 
>     7. symlink binaries
> 
> 
> ```
> sudo ln -s ~/wine/bin/wine /usr/local/bin/wine
> sudo ln -s ~/wine/bin/wineboot /usr/local/bin/wineboot
> sudo ln -s ~/wine/bin/winecfg /usr/local/bin/winecfg
> sudo ln -s ~/wine/bin/wineserver /usr/local/bin/wineserver
> sudo chmod +x /usr/local/bin/wine /usr/local/bin/wineboot /usr/local/bin/winecfg /usr/local/bin/wineserver
> ```
> 
> now you should be able to install rain world by running `wine rain_world_gog_installer.exe` _Note. the name is likely diffrent_
> 
> after installation, the rain world should be located at `~/.wine/drive_c/GOG Games/Rain World`
> 
> (inside that directory) you can run it with
> 
> ```
> BOX86_DYNAREC_STRONGMEM=1 wine RainWorld
> ```
> 
> _Note. BOX86_DYNAREC_STRONGMEM prevents game from crashing for no reason_ _Note. on devices with mali gpu adding `MESA_GL_VERSION_OVERRIDE=3.2 PAN_MESA_DEBUG=gl3` **might** make it possible to run the game_
> 
> Tested on lenovo ideapad duet 5 (homestar)
> 
> Based on [this](https://forum.armbian.com/topic/19526-how-to-install-box86-box64-wine32-wine64-winetricks-on-arm64/) working on op6t is a bit overwhelming hence the tutorial for me to chill out

The link off hangover is dead

thenameisluk, 2025-01-10:
fixed

archisman-panigrahi, 2025-09-06:
Can we move/copy these guides to the website for increased visibility?

thenameisluk, 2025-09-26:
works for me