velvet-os/imagebuilder #282: problem: usb is unreliable and with bad performance on at least kukui and trogdor

hexdump0815, 2025-01-14:
using usb devices on kukui and trogdor devices is sometimes unreliable (for instance they cannot be used to connect a suzyqable to another chromebook, sometimes usb audio gives trouble etc.) and the io performance when using at least storage devices connected via usb on those devices can be very poor. this issue is here to track this problem, its details, its analysis and the tries or ideas to solve it. please add discoverd problems and devices they were discovered on with kernel versions used below as well as well as ideas and experiments to find out where the problem comes from.

ideas:
- verify that those performance and reliability problems do not exist in chromeos / with a chromeos based kernel
- for trogdor i'll check if i see the same problem on the samsung galaxy book go which is qcom sc7180 based as well, but not a chromebook (woa instead)

thenameisluk, 2025-01-14:
i've done some testing
previously assumed flash storage is faster that usb hdd drive (apparently not)

the situation is not as bad as i thought and different is around 50%

with the same task of dumping partition from the device, the write speed (on the same setup) was

sh1mmer 150 mb/s
[image]

velvet os 100 mb/s (unstable between 90 - 110 mb/s)
[image]

which isn't that bad, still the broken suzyqcable and other devices (my CD readers) is issue that should be addressed

thenameisluk, 2025-01-16:
got some new info
on mainline stb-cbm corsola also loses ability to work with suzyq cable, which means like the same issue occures

that implies that there is something missing/broken in mainline but not inside colabora/chromeos kernel

hexdump0815, 2025-01-16:
good to know - looks like something special about usb on those mediatek devices has not been mainlined yet ... i plan to ask in the #linux-mediatek irc channel at some point as there are some people around who are doing the mediatek mainlining - maybe they have an idea if we can describe the problem clear enough ... thanks a lot for testing it on corsola