velvet-os/imagebuilder #287: Add veyron-tiger to the list for build/testing?

dragon788, 2025-01-23:
I have several of these devices (and the Chromebit CS10) and I might try testing with the generic veyron image, but I wanted to see what the probability of it working was before going crazy, as I've had success with postmarketOS and the SXMO UI, but it has been a bit since I did those.

https://github.com/alpernebbi/depthcharge-tools/blob/master/depthcharge_tools/boards.ini#L2457

thenameisluk, 2025-01-23:
since there are no notes on tiger (i could find)
i assume we have no tiger hardware to test if it will work
you need to try it yourself

looking at uboot notes
https://github.com/hexdump0815/u-boot-chainloading-for-arm-chromebooks/blob/master/readme.cbr
it isn't listed as potentially supported

but it might work

hexdump0815, 2025-01-24:
i can only agree to what @thenameisluk already said: i did not play around with tiger myself and i'm not aware of anyone having tried it ... if you are willing to experiment a bit i would say the chance is there - maybe start with trying to boot a kernel kpart image first before trying to get u-boot working ... at least there is a dts for tiger there in mainline, so it might just work this way ...