velvet-os/imagebuilder #291: Touchpad not working and no network devices are shown

jramirez281, 2025-02-02:
Hello,
I am a noob so please bear with me. I booted the kukkui 230917 image from a usb on my chromebook (Lenovo ideapad 3 Chrome 14M386) and followed [these](https://github.com/hexdump0815/imagebuilder/blob/main/doc/chromebooks/installation/basic-installation.md) instructions. 

I was able to boot into debian without my usb being in the port, but my touchpad was not responsive at all, and none of my network devices showed up. I was not able to connect to the internet, and typing ip a only showed me an entry for lo, no wlan0. Everything was fine when I booted from the USB, this only happened after installing and booting from the ssd. 

Thank you

thenameisluk, 2025-02-02:
@jramirez281 if things work on usb system they should as well work on internal drive

could you try repeating the installation making sure you've read/done everything correctly?

jramirez281, 2025-02-02:
Sure, I just have a question about step 3. The usb is under sda device, so should I input: 
`export srcPart=sda`
or 
`export srcPart=mmcblk0p`? When I boot from the usb, is the usb the booted device? I am a bit confused. Thank you

jramirez281, 2025-02-02:
I ask this because I inputted 
`export srcPart=sda` the first time I tried to do this (I've attempted this multiple times) and I was unable to boot properly unless I plugged in the usb drive. I suspected this line to be the cause of the problem, and so this time I inputted `export srcPart=mmcblk0p`, but I really have no idea.

hexdump0815, 2025-02-02:
"export srcPart=sda" should be correct in your case

thenameisluk, 2025-02-02:
the booted device is the one with ```/``` in mount point section when you run ```lsblk```

could you be more specific than "I was unable to boot properly"?
without details i can only speculate what could be the issue
a picture of the screen you are getting might be useful

jramirez281, 2025-02-02:
Haha sorry. I was unable to boot properly as in the screen with the text showed up but I could'nt get past it i.e. to the login page, unless I had the flash drive plugged in. It was giving some sort of error. I can't replicate it now since I redid the install, but I can send pictures of whatever you need once I get home.

jramirez281, 2025-02-02:
Also really appreciate you both taking time out of your sunday to help me out.

thenameisluk, 2025-02-02:
if ur device doesn't boot properly you should send picture of the screen
it usually has some important details about what went wrong
there are a few known ways chromebook fails to boot so i don't know which one was the case this time

jramirez281, 2025-02-03:
got it, will try to repeat the installation

jramirez281, 2025-02-03:
I am unable to boot from usb anymore. It keeps giving me the messages shown in the picture. Let me know if it is too small, I was just trying to get as much as I could in frame.

[image]

jramirez281, 2025-02-03:
ok was able to boot into debian from usb, but when I try to follow the instructions I get these errors.

[image]

thenameisluk, 2025-02-03:
the errors on the first image mean usb got disconnected while system was running so it's not mounted anymore
second one probably the same

also i wouldn't proceed with the installation without lsblk

it's likely system on the usb got corrupted after being disconnected, you should probably reflash it

hexdump0815, 2025-02-03:
@jramirez281 - maybe try another usb device for booting - there are lots of fake or broken usb sticks around ... might not be it, but definitely worth a try

jramirez281, 2025-02-04:
So I write the image onto another usb disk and successfully booted from it. i was following the steps but reached a hiccup at step 8. I received the error in the photo.

[image]

thenameisluk, 2025-02-04:
@jramirez281 u need to append -f command as it tells you to do
also remember to gain root and export the disks again if you've closed this terminal

thenameisluk, 2025-02-04:
also unmount the mounted partitions
you can ```lsblk``` to see the mount point and ```umount <mount point>```

don't unmount usb drive only internal one

jramirez281, 2025-02-04:
I booted from my usb again but the touchpad no longer works unfortunately. I'm not sure what to do.

thenameisluk, 2025-02-04:
there is a note [here](https://github.com/hexdump0815/imagebuilder/blob/main/systems/chromebook_kukui/readme.md)
"juniper: sometimes the touchpad does not seem to work after booting, rebooting usually helps"
not for your specyfic chromebook (other kukui device) but might be same issue

check if rebooting works

jramirez281, 2025-02-05:
I rebooted the system and the exact same thing happened...

jramirez281, 2025-02-05:
Ok I finished the steps and everything seems to be ok. Looks like even the touchpad and network interface is working. Thank you both!

jramirez281, 2025-02-05:
Yes I skipped the optional step. I was in the middle of it just now and I got the error in the picture. Should I skip this particular command?

[image]

thenameisluk, 2025-02-05:
could you try running ```mkdir /swap```
and running the command again

jramirez281, 2025-02-05:
This is what I got:

[image]

jramirez281, 2025-02-05:
Cd'd into the directory and inputted the last command. Got this error:

[image]

thenameisluk, 2025-02-05:
well it partially worked
you have swap but it is compressed
so should work

jramirez281, 2025-02-05:
ok great! This is what I got when I rebooted the system and inputted the systemctl command again:

[image]

jramirez281, 2025-02-05:
Thanks for all the help!

thenameisluk, 2025-02-05:
no problem