velvet-os/imagebuilder #292: WiFi dosen't show up on duet 5

votisek, 2025-02-02:
I just installed ubuntu jammy with the prebuilt image in releases on my Lenovo Duet 5 (homestar). The touchscreen also dosen't work. Would love some help. Thanks.

thenameisluk, 2025-02-02:
@votisek
wifi is a known ubuntu issue https://github.com/hexdump0815/imagebuilder/blob/main/doc/important-information.md
can be fixed like this https://github.com/hexdump0815/imagebuilder/issues/231#issuecomment-2212464410

about the touch screen
you need to update the kernel
6.6.23
https://github.com/hexdump0815/linux-mainline-and-mali-generic-stable-kernel/releases/tag/6.6.23-stb-cbq%2B
or if you want something newer
6.12
https://github.com/hexdump0815/imagebuilder/discussions/267#discussioncomment-11289341
the issue should go away

also there if you want to get audio comming out of all 4 built-in speakers instead of 2
https://github.com/hexdump0815/imagebuilder/discussions/240#discussioncomment-10732006

votisek, 2025-02-02:
@thenameisluk 
There is a problem, I need internet access to apt install stuff, so, is there another way, or do i need to somehow connect my duet to ethernet?

votisek, 2025-02-02:
its allright now, found a kernel-compatible wifi usb dongle.

votisek, 2025-02-03:
@thenameisluk  
after reinstalling to debian and some debugging i have successfully started using it normally, but the screen lags alot and it seems random. After the spike some of the cpu cores spike in value to almost 100%. I am really confused by this because sometimes, i have to close and open my laptop, but sometimes its really short and fixes itself right away. So far I havent encountered any other issues while using it practically and really like it.

thenameisluk, 2025-02-03:
weird
i've never encountered any random lag spikes on my duet 5
've experienced some when i overloaded cpu while compiling on all 8 cores but never for no reason

unless you are using kde which i remember had that issue on one of my chromebooks, i think it is doing synchronous read from a disk which locks the other threads, but that's just a theory

if you can could you run ```dmesg``` command after a lag spike to see if there is anything interesting that would be great

also glad you like it, duet 5 is my favourite chromebook and trusty daily driver

votisek, 2025-02-03:
Allright, i am going to sleep now, but will update about the issues and fixes, thanks. But i must ask, what browser you use? I did some testing and the lagspike is always with the default firefox open.

hexdump0815, 2025-02-03:
@votisek - how much memory does your duet 5 have? if only 4gb and you use a lot or complex tabs in firefox it might run out of memory and the velvet os images are configured to use zswap, i.e. a part of the memory is put aside and then everything in it gets transparently compressed/decompressed on access - this helps to kind of provide a bit more memory than there really is and its still faster than paging/swapping out to disk (also that if needed is more efficient this way as only the compressed pages = much less - get paged out to disk). i have written a small script to get an idea of how much that memory compression is used: /scripts/zswap-status.sh

an example of my freshly booted kukui kappa (4gb ram) with a firefox with dozens of tabs open:
```
root@kappa:~# /scripts/zswap-status.sh 
Zswap enabled:            Y
Percentage:               25 - i'm reserving 25% of the memory to be used compressed (default of the images is 20%)
Compressor:               zstd
Zpool:                    zsmalloc

Page size:                4096
Stored pages:             465351
Pool size:                432Mi - currently that much memory is used for compressed pages
Decompressed size:        1.8Gi - this is the amount of uncompressed memory the represent
Written to storage size:  1.1Gi - this is how much is really written to storage (as with zswap what you see with free, top etc. is not correct)
Page compression ratio:   4.210 - that it the compression rate achieved for the compressed memory, factor 4+ really is something
Total real mem size:      3.8Gi - physical ram of the machine
Total virtual mem size:   5.2Gi - this is the physical ram without compression (75% in my case) plus the uncompressed size of the compressed memory pages, so my machine now has more than 5gb of ram virtually :)
```
this works very well as long as there is still enough memory in the system so that it is not all the time busy compressing and decompressing pages (which is fast with modern cpu cores but way slower than accessing memory directly but still faster than using swap on disk) but can result in spikes as you described if there simply is not enough real ram for the kind of usage scenario of the system as a lot of cpu time is used for compression/decompression and in the end maybe even paging to disk ... its still better than just disk swap, but has its limit

besides that it requires about 1.5x the physical ram as swap size (see /scripts/recreate-swap.sh for recreating it with a different size - i'm using 6gb on my kappa for instance) to work properly as linux counts the compressed pages in memory and the ones really written to disk as swapped out in commands like free, top etc. and as such needs more swap space although not as much as it seems is really used on disk

votisek, 2025-02-04:
Thank you hexdump for the info, unfortunately i have 8gb of ram and 256gb of emmc storage. On the bright side i just got the 'dmesg' output and noticed thois at the end: 
'''[  300.687385] systemd-journald[322]: Time jumped backwards, rotating.
[  305.414174] systemd-journald[322]: Time jumped backwards, rotating.
'''

thenameisluk, 2025-02-04:
that one just means cpu was too overloaded even for systemd to work

votisek, 2025-02-04:
uninstalling firefox and using chromium helped, i got no lag whatsoever from the point of closing firefox. thank you for all the help and I must say, this repo is really good, ive wanted to use linux on my duet since i bought it and had no hope until i found this repo. Thank you so much for this being a thing and for luk and hexdump for helping me.

closed 2025-02-04 by votisek

thenameisluk, 2025-02-04:
no problem