velvet-os/imagebuilder #294: Asus cm1402c dbus error after login

ereish64, 2025-02-03:
I have successfuly booted my Asus CM1402c using the image provided at
https://github.com/hexdump0815/imagebuilder/releases/tag/240112-01

[image]

I login with the default credentials:
Uname: linux
Passwd: changeme

After a while I get an error that when closed returns me back to the login screen yet again:
[image]

A device status report mentioned that this could be fixed by running `sudo apt install dbus-x11`
#49 

I cannot for the life of me get to the terminal. I've tried CTRL+ALT+F2 as well as the other Fx keys. CTRL+ALT+T also had no effect.

Any input is appreciated.

thenameisluk, 2025-02-04:
@ereish64 
you need an external keyboard to get into tty
the chromebook keys do literally what the image on them shows (not F1-F12) but you can remap them
https://github.com/hexdump0815/imagebuilder/blob/main/doc/postinst/modifying-keyboard.md

also are you sure your device is corsola?
could you provide a link to ur device specification with it's cpu specified?

ereish64, 2025-02-04:
I had no idea those keys still did the same thing outside chromeos. I'll give that a try when I get home.

As for the device, it's a ASUS CM1402cm2a 4gb variant. Full specs [HERE](https://www.asus.com/laptops/for-home/chromebook/asus-chromebook-cm14-cm1402c/techspec/?awc=31828_1738671459_ceac43a76c4309bb665cd9de91b71d6e)

thenameisluk, 2025-02-04:
well it does look like corsola
weird it wasn't listed in pmos

ereish64, 2025-02-05:
I reflashed the OS on a MicroSD card this time and put it directly into the reader of the PC. That way I could use the USB port for a keyboard. The PC had no issues booting to the desktop.

My only guess is that the SD card reader I was using before was so slow that dbus was timing out or something of the sort. Very strange.

I really appreciate the help with this oddity and I love what you guys are doing here! You turned my slow chromebook into a usable PC!

closed 2025-02-05 by ereish64

thenameisluk, 2025-02-05:
read speed has no affect on that
ur usb might have been reporting fake capacity
look at https://www.alivestudios.co.uk/usb-stick-scam/