velvet-os/imagebuilder #296: WiFi not working in Ubuntu Noble image for Chromebook Oak (hana)

archisman-panigrahi, 2025-02-13:
I am using a Lenovo 300e Chromebook (hana/oak).
It works fine with the Ubuntu Jammy and Debian Bookworm images available [on this repo](https://github.com/hexdump0815/imagebuilder/blob/main/systems/chromebook_oak/readme.md).

I built an Ubuntu Noble image (I am using the default kernel supplied by imagebuilder 6.6.9) using the imagebuilder framework.

While the image boots, it cannot find any wifi adapters. WiFi worked well in the Jammy and Bookworm images.

How to debug this?

thenameisluk, 2025-02-13:
@archisman-panigrahi does the wifi work on older kernel?

archisman-panigrahi, 2025-02-13:
Now that I have an installed system, how do I change the kernel to test this?

archisman-panigrahi, 2025-02-13:
I copied the contents to `/usr/lib/firmware` from my working Debian bookworm system to Noble and now wifi works.

Does anyone know which particular file is responsible for the wifi? Then the imagebuilder script will be able to take care of this

thenameisluk, 2025-02-13:
weird oak doesn't seam to have any additional firmware
https://github.com/hexdump0815/imagebuilder/tree/main/systems/chromebook_oak/extra-files/lib

unlike trogdor https://github.com/hexdump0815/imagebuilder/tree/main/systems/chromebook_trogdor/extra-files/lib

can you try installing
```firmware-linux``` package and seeing if it works?
maybe it's just included in there
https://packages.debian.org/trixie/firmware-linux

archisman-panigrahi, 2025-02-13:
My current Noble setup already has working wifi after copying the lib/firmware files.

Are you suggesting I start with a fresh image, connect to internet via USB tethering, and install `firmware-linux`, and check if wifi works?

thenameisluk, 2025-02-13:
you can if you want
it would be nice to check if it fixes the issue

archisman-panigrahi, 2025-02-14:
Ubuntu does not have the `firmware-linux` package but it has `linux-firmware`, which was already installed (so, it does not fix wifi).

Also, the performance is really bad compared to bookworm. In bookworm, everything works like a charm and there is no lag. In Noble, when I run APT, other software are not properly responding (they don't receive keystrokes, and when I click on some button it does not register). Also, when I open Chromium, simple sites like linuxmint.com are taking forever to load and are freezing the OS.

The interesting thing is, the cpu load is always under 25-30% (according to htop), but for some reason,  it seems as if the computer is still freezing (since running apps are freezing (they are in RAM), it could not be attributed to a slow microSD card).

Any idea how I can debug this?

This issue does not happen in Debian bookworm at all.

Let me check how it works on Jammy and Trixie.

thenameisluk, 2025-02-14:
maybe ur cpu is capped at 25%
so the device is on 1/4 of its regular speed

archisman-panigrahi, 2025-02-14:
> maybe ur cpu is capped at 25%
> so the device is on 1/4 of its regular speed

But that didn't happen in bookworm. Let me double check and get back.

hexdump0815, 2025-02-14:
@archisman-panigrahi - i have no idea what is going on with noble on that system ... sadly our time is limited and we are focussing currently on preparing the next release based on debian trixie and only when this is out (still quite a few weeks to go there) we might have a closer look at ubuntu noble ... i can only say that i did a quick test of noble on an oak-elm system and from what i remembered it worked normal ... could you try the new unified mediatek kernel either from here: https://github.com/hexdump0815/linux-mainline-mediatek-mt81xx-kernel/releases/tag/6.12.12-stb-cbm%2B or from the debian repo of @thenameisluk - he might give more detailed instructions how to use that

closed 2025-06-15 by archisman-panigrahi