velvet-os/imagebuilder #297: <system>: status: <device>

94alshamsi94, 2025-02-16:
Notes on <device> running <kernel version>.

bro we need driver for rtl8811au because custom header i can't build my copy is oak on lenovo chromebook 100e (mt8173 version) - [hana](https://github.com/hexdump0815/imagebuilder/blob/main/doc/chromebooks/systems/oak/hana-100e-gen2.md)
lenovo chromebook 300e (mt8173 version) - hana
lenovo chromebook n23 - hana

thenameisluk, 2025-02-16:
this template isn't meant to be used this way
just use a blank issue and fix this title

if you need kernel headers here are the instructions
https://github.com/hexdump0815/imagebuilder/blob/main/doc/postinst/installing-kernel-headers.md

archisman-panigrahi, 2025-02-27:
The wifi works out of the box except in Ubuntu 24.04 (noble). Which OS are you running?

thenameisluk, 2025-02-27:
it also works on debian

the driver requested is for realtek wifi card

while at least [100e uses Marvell one](https://psref.lenovo.com/syspool/Sys/PDF/Lenovo/Lenovo_100e_Chromebook_2nd_Gen_MTK/Lenovo_100e_Chromebook_2nd_Gen_MTK_Spec.PDF)

so i'm assuming it's for usb dongle or similar