velvet-os/imagebuilder #304: How to enable usb boot in google scarlet (dru)

archisman-panigrahi, 2025-02-25:
I enabled developer mode in google scarlet (similar to dru), Acer Chromebook Tab 10.
It refuses to boot because usb_boot is mnot turn on. How to turn it on? Unlike a chromebook laptops, I cannot open the console on the tablet.

thenameisluk, 2025-02-25:
@archisman-panigrahi can't you plug in external keyboard over usb-c?
it should work

thenameisluk, 2025-02-25:
alternatively

boot into chromeos

go into browser an open
```
chrome-untrusted://crosh
```

then type
```
shell
```
in console

and enter
```
sudo crossystem dev_boot_usb=1 dev_boot_signed_only=0
```
in shell

archisman-panigrahi, 2025-02-25:
> alternatively
> 
> boot into chromeos
> 
> go into browser an open
> ```
> chrome-untrusted://crosh
> ```
> 
> then type
> ```
> shell
> ```
> in console
> 
> and enter
> ```
> sudo crossystem dev_boot_usb=1 dev_boot_signed_only=0
> ```
> in shell

That worked! Thank you. Can we add this to the readme in the tablets section?

It has a section on "Booting from usb (or sd)", which says how to enable usb_boot. Since tablets don't always have a keyboard, adding this information about how to open shell in that place in the readme will be helpful.

archisman-panigrahi, 2025-02-25:
More specifically, this readme https://github.com/hexdump0815/imagebuilder/blob/main/doc%2Fchromebooks%2Freadme.md

closed 2025-02-25 by archisman-panigrahi